realsenseai/librealsense #4939: a plane is measured as a concavity

Cavendish-Koo, 2019-09-26:
| Camera Model                       | {D435} | 

I have recently encountered a problem. A plane is measured by the camera as a concave surface.
[image]
As is shown above, The depth value in the middle of the plane is lower than the two sides.I have recalibrated the camera, and things get a little better.But the depth value in the middle of the plane is still 10-20 cm lower than the two sides. I hope somebody can help me and reply me  quickly..

MartyG-RealSense, 2019-09-26:
The way the image 'bows' in at the center reminds me of a distortion phenomenon that is rare (at least in Librealsense) called 'pincushion distortion' (the opposite of barrel distortion, which bulges outward).  I have not seen a reported case of it in Librealsense since the old 1.12 legacy SDK in 2016.

[image]

Here's the case that was referenced:

https://github.com/IntelRealSense/librealsense/issues/61#issuecomment-194380453

Hopefully, if it is a pincushion distortion, a member of the Intel RealSense team on this forum can provide helpful insight about how to correct it.

Cavendish-Koo, 2019-09-26:
Thank you for the quick reply.
I have five realsense d435 camera and only one camera doesn't have the distortion. Haha, maybe this phenomenon is not rare for me.  And the depth value in the middle is lower than two sides. Is it a barrel distortion rather than pincushion distortion?

MartyG-RealSense, 2019-09-26:
A barrel distortion bulges outwards instead of inward.

[image]

Actually, you have a point.  If the depth reading at the center is lower, this would suggest that the pixels are nearer to the camera than the ones at the side, indicating a bulge rather than a depression (where you might expect the pixels to be further away and have a higher depth value).  Apologies for that!

Cavendish-Koo, 2019-09-26:
So how can I solve this problem. Can I only return to the factory for repair? I have a more urgent project, and long-term maintenance is unbearable.

MartyG-RealSense, 2019-09-26:
If cameras are within the return period (30 days from purchase I believe) then they can be returned to Intel (if you bought it from them) or from the retailer that you bought it from.  There is only refund or replacement, as Intel do not have a repair facility for RealSense cameras.

Could you check what temperature your cameras are running at please?  This can be done in the RealSense Viewer.  For a cold startup, I would expect to see ASIC temperature values of mid to late 20's.  If your cameras have temperatures of 40 or more from cold startup, that would suggest a problem.

dorodnic, 2019-09-26:
I think Dynamic Calibrator software *may* help. Depth image is synthetic and is generated using extrinsic calibration between left and right sensors. In our experience relative rotation between the two sensors can create concave / convex depth. @agrunnet - what is your opinion?

RealSenseCustomerSupport, 2019-10-15:
----------------------------------------------

Please let us know if you need further assistance about this topic. Thank you.

closed 2019-10-16 by Cavendish-Koo

octalframes, 2019-11-27:
Even after dynamic calibration we are facing pincushion distortion!

tispratik, 2020-03-14:
We are also seeing heights in the center to be lower than at the fringes. Our camera is mounted 24" from a flat conveyor belt. We measure the heights of objects by taking a difference of empty conveyor distance measured minus distance measured to the center of the object. We see a difference of 2cm (0.02m) between the center and the fringes. If the height of an object observed in the center is 10cm, on the fringes it shows up as 12cm.

octalframes, 2020-03-15:
And they no solution for this or they dont want to give!

Cavendish-Koo, 2020-03-15:
As a depth camera, the D400series is not satisfactory in terms of measurement accuracy. You can't restore the geometry of objects well on point clouds. However I can give some of my previous solutions. It is to fit these errors with a function. Although this method is not very good, it is barely enough. Hope this helps you.

octalframes, 2020-03-15:
Accuracy is one thing. Distortion(Barrel, Pincushion) is another. Intel should make software to rectify distortion dependent on intrinsic values open source. That will be a step truly towards open source and in interest of developer community. After this experience we are hesitant to try out any Intel products be it Lidar Cameras, open vino etc..

tispratik, 2020-03-15:
@Cavendish-Koo we were actually looking to develop similar function, it will be great if you can share your solution. Thank you!

Cavendish-Koo, 2020-03-16:
I found that the distortion of a point on the depth image is mainly related to the pixel distance from the center pixel of the image and the distance from the camera to the target. 
I first placed the camera parallel to a plane, then acquired several depth images at different distances and recorded the true distance. This way we can get some triples (pixelCol, depthValue, realDistance). Then I used matlab to fit this surface.
 In this way, I can get realDistance according to pixelCol and depthValue. This method ignores vertical distortions. Hope this will help you.

Cavendish-Koo, 2020-03-16:
@octalframes That makes sense. But as developers, all we can do is feed back problems to manufacturers and think about possible solutions. I hope both manufacturers and developers can do better and better.