realsenseai/librealsense #4942: How to synchronize hardware timestamp?

shadowdk3, 2019-09-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |   Linux16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-60                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.28.1                        | 
| Language                            |  cpp                         | 
| Segment			| others                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I setup two D435 in order to sync two video together. I am confuse why synchronization fail.
I did the following steps:

1. Hardware sync two camera? I connect two cameras with cable, pin5 to other, pin9 to other
2. set `RS2_OPTION_INTER_CAM_SYNC_MODE` 1 to master and 2 to slave
3. set `RS2_OPTION_AUTO_EXPOSURE_PRIORITY` to 0
4. resolution 640 * 480 @ 30fps

I try to get system timestamp and hardware timestamp from depth frame, I find hardware timestamp has different between two cameras, even wait for 10s.
Also, I have a few more question want to clarify.

1. Does system timestamp equal to global timestamp?
2. Which hardware clock should be synchronize when two cameras synchronize? `SENSOR_TIMESTAMP` or `FRAME_TIMESTAMP`?
3. Can two RGB image synchronize now?

Thank you very much.

Following is my result
system clock get from `BACKEND_TIMESTAMP`
hardware clock get from `FRAME_TIMESTAMP`

device: 0 system clock: 1569491356102 hardware clock: 239500271 

device: 1 system clock: 1569491356135 hardware clock: 240009648 

device: 0 system clock: 1569491356135 hardware clock: 239533626 

device: 1 system clock: 1569491356168 hardware clock: 240043003 

device: 0 system clock: 1569491356168 hardware clock: 239566981 

device: 0 system clock: 1569491356203 hardware clock: 239600336 

device: 1 system clock: 1569491356203 hardware clock: 240076358 

device: 0 system clock: 1569491356235 hardware clock: 239633691 

device: 1 system clock: 1569491356235 hardware clock: 240109713 

device: 0 system clock: 1569491356268 hardware clock: 239667046 

device: 1 system clock: 1569491356268 hardware clock: 240143068 

device: 0 system clock: 1569491356303 hardware clock: 239700401 

device: 1 system clock: 1569491356303 hardware clock: 240176423 

device: 0 system clock: 1569491356335 hardware clock: 239733756 

device: 1 system clock: 1569491356335 hardware clock: 240209778 

device: 0 system clock: 1569491356368 hardware clock: 239767111 

device: 1 system clock: 1569491356369 hardware clock: 240243133 

device: 0 system clock: 1569491356402 hardware clock: 239800466 

device: 1 system clock: 1569491356402 hardware clock: 240276488 

device: 0 system clock: 1569491356435 hardware clock: 239833821 

device: 1 system clock: 1569491356435 hardware clock: 240309842 

device: 0 system clock: 1569491356469 hardware clock: 239867176 

device: 1 system clock: 1569491356469 hardware clock: 240343197 

device: 0 system clock: 1569491356502 hardware clock: 239900531 

device: 1 system clock: 1569491356502 hardware clock: 240376552 

device: 0 system clock: 1569491356535 hardware clock: 239933885 

device: 1 system clock: 1569491356535 hardware clock: 240409907 

device: 0 system clock: 1569491356569 hardware clock: 239967240 

device: 1 system clock: 1569491356569 hardware clock: 240443262 

device: 0 system clock: 1569491356602 hardware clock: 240000595 

device: 1 system clock: 1569491356602 hardware clock: 240476617 

device: 0 system clock: 1569491356635 hardware clock: 240033950 

device: 1 system clock: 1569491356635 hardware clock: 240509972 

device: 0 system clock: 1569491356669 hardware clock: 240067305 

device: 1 system clock: 1569491356669 hardware clock: 240543327 

device: 0 system clock: 1569491356702 hardware clock: 240100660 

device: 1 system clock: 1569491356702 hardware clock: 240576682 

device: 0 system clock: 1569491356735 hardware clock: 240134015 

device: 1 system clock: 1569491356735 hardware clock: 240610037 

device: 0 system clock: 1569491356770 hardware clock: 240167370 

device: 1 system clock: 1569491356770 hardware clock: 240643392 

device: 0 system clock: 1569491356802 hardware clock: 240200725

dorodnic, 2019-09-29:
What is the value of `frame.get_timestamp_domain()` in your system? 
The way this is intended to be solved is for timestamp domain to report `GLOBAL_TIME` and `get_timestamp` to report values consistent across all devices.

shadowdk3, 2019-10-01:
@dorodnic sorry for late reply, I get `GLOBAL_TIME` when I run `frame.get_timestamp_domain()`.
I notice that two camera seem to be sync for 0.1ms, but when the motion is too fast, two video cannot be sync.

dorodnic, 2019-10-02:
> when the motion is too fast, two video cannot be sync

Are you referring to the RGB or depth/IR streams?

shadowdk3, 2019-10-02:
@dorodnic Depth stream

dorodnic, 2019-10-02:
I don't want to duplicate information available [here](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration)

How do you measure that the depth stream is not synchronized?

shadowdk3, 2019-10-03:
we let a ball free fall, we capture both color and depth image. we get different ball level in two cameras. ball level difference more and more when time is getting longer

RealSenseCustomerSupport, 2019-10-28:
----------------------------------------------

Hi, please let us know if  you can capture synchronized frames and timestamps or still have the same problem?
Thank you.

closed 2019-11-01 by RealSenseCustomerSupport