realsenseai/librealsense #4943: How to get Left AND Right color camera images from Intel RealSense

awatson3, 2019-09-26:
Using the LibRealSense API, is it possible to simultaneously stream color images from both the left **and** right cameras on *any* of the Intel RealSense hardware?

MartyG-RealSense, 2019-09-26:
I recently asked Intel about getting a color stream from the left and right IR imagers of the 400 Series cameras.  The response given is in the link below.

https://github.com/IntelRealSense/librealsense/issues/3754#issuecomment-487367179

awatson3, 2019-09-26:
@MartyG-RealSense Thank you

@RealSenseCustomerSupport Per above comments, there are no Realsense models that support simultaneous RGB streams from Left and Right cameras, correct?

dorodnic, 2019-09-29:
Yes, there is no such configuration at the moment.

RealSenseCustomerSupport, 2019-10-14:
----------------------------------------------

@ awatson3 Do you have further questions about this topic?

closed 2019-10-24 by RealSenseCustomerSupport