realsenseai/librealsense #4946: FRAMOS Depth Camera D435e Starter kit

tonyhsiao1995, 2019-09-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { FRAMOS D435e } | 
| Firmware Version                   | 5.11 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-60-generic   | 
| Platform                                 | PC  |
| SDK Version                            |  2.20                         | 

### Issue Description
I am using FRAMOS D435e Starter Kit. After installation, my realsense-viewer only can use D435e with fps 30 in 1280 x 720 resolution. I can't change to 15 or 8 fps.

Marenix, 2019-09-27:
FRAMOS D435e currently supports only resolution 1280x720@30 fps.

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@tonyhsiao1995
Do you have further question on this?

tonyhsiao1995, 2019-10-12:
No, thanks

closed 2019-10-12 by tonyhsiao1995