realsenseai/librealsense #4947: SR305 : cannot stream depth @ 60fps

blackccpie, 2019-09-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.28.1                          | 
| Language                            |  RealSense Viewer (C++)                          | 
| Segment			|  others                           | 

### Issue Description

I've started testing the new SR305 with the Realsense Viewer, but when I activate the depth stream with the 640x480@60fps mode, the real measured framerate given in the info section of the stream only reaches 50fps.

I don't think the problem is related with my machine, on which I regularly use D4XX's 60fps modes wihout problem.

Is it a normal limitation or am I doing something wrong?

Thx for your help

MartyG-RealSense, 2019-09-27:
I located one previous case in which an SR300 user had their stream capped at 50 FPS at most when running in 60 FPS mode (the SR305 is essentially an SR300 camera inside a new casing).  Please see the comment linked to below, and the response given in the comment below it.

https://github.com/IntelRealSense/librealsense/issues/802#issuecomment-343750746

blackccpie, 2019-09-27:
@MartyG-RealSense : concerning the firmware version, that seems quite strange to perform a downgrade to get better performance... in the homepage it is stated `3.10.10.0 +`, which corresponds to a minimal version.
By the way the 3.10.10.0 firmware is nowhere to be found as a standalone download usable in the Viewer update menu (neither is the 3.27.3.0 if I want to rollback). The only tool I found is the [DCM](https://downloadcenter.intel.com/fr/download/25044/Intel-RealSense-Depth-Camera-Manager-DCM-?elq_cid=3361846).

I also performed some tests on the color stream, and I also have fps issues when activating the 640x480@60fps mode (fps is around 30-35fps).

Could you confirm that you have normal framerates on your win10 machines?

Regards

MartyG-RealSense, 2019-09-27:
My memory of historical events from some years ago was that the 3.27 firmware did indeed come after 3.10 but there were problems after that.  So it makes sense to recommend 3.10 as a potentially more stable firmware.

I believe that 3.10 is not available as a standalone firmware file and that the only way to currently install it is through the RealSense Viewer.

I have not personally noticed frame-rate issues on Windows 10 with my SR300 (I do not have access to an SR305).  When problems do occur (e.g disconnections), I have found that reducing the value of the Laser Power setting increases stability for me personally.

blackccpie, 2019-10-02:
I tried to install the recommended firmware using the Realsense Viewer, and that didn't solve the issue. 
I also tried the camera using my Ubuntu 18.04 partition, and I'm also stuck @50fps. 
The 60fps feature is the reason why I bought this camera, so solving this issue is quite important for me.

MartyG-RealSense, 2019-10-02:
I ran some tests with my SR300 using your settings and found I was also getting 50 FPS on the 60 FPS mode in the RealSense Viewer.  Further tests found that there was a linear relationship between the FPS and a setting under the 'Controls' oprions called "Motion Range".  Increasing the motion range value reduced the FPS, and reducing the value increased FPS.  When motion range was zero, FPS was 60.

[image]

dorodnic, 2019-10-02:
Hi @blackccpie 
@MartyG-RealSense beat me to it, but yes, Motion-Range control is equivalent to IR exposure and can reduce actual FPS.

blackccpie, 2019-10-03:
Hi @MartyG-RealSense & @dorodnic, that did the trick indeed, when reducing Motion-Range value I could increase the framerate up to 60fps.
I never had this issue with sterescopy based cameras like the D4XX, is this control specific to the SR305?

Regards

MartyG-RealSense, 2019-10-03:
Apparently it is an SR3x specific issue - those cameras are based on coded light / structured light instead of stereo.

https://github.com/IntelRealSense/librealsense/issues/3185#issuecomment-460570599

closed 2019-10-03 by blackccpie