realsenseai/librealsense #4948: depth quality tool crashes

sodom89, 2019-09-27:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows10|
|**Name**|Intel RealSense D435|
|**Serial Number**|831612070780|
|**Firmware Version**|05.11.15.00|

If I'm trying to record the metrics with the depth quality tool. It crashes while closing the window and not providing any error code. The I got the same behaviour with den Release 2.28.0. It seems to be independent from the settings of fps, resolution and region of interest. Do you have any suggestions?

antoan, 2019-09-27:
Name | Intel RealSense D435i
Firmware Version | 05.11.11.100
OS Ubuntu 16.04

Same here. Crash occurs on initiating recording.

ev-mp, 2019-10-02:
This is a known bug in the SDK.
Duplicate of #4913

closed 2020-03-18 by RealSenseCustomerSupport