realsenseai/librealsense #4949: Assertion failure when running python tutorial

sanjanasrivastava, 2019-09-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | { 5.11.6.250 } |
| Operating System & Version |   { MacOS Mojave } | 
| Platform                                 | { PC } |
| SDK Version                            |  { 2.28.1 }                          | 
| Language                            |  { python }                          | 
| Segment			|  { others }                          | 

### Issue Description

I am attempting to get started with pyrealsense2. I tried running `librealsense/wrappers/python/examples/python-tutorial-1-depth.py` directly from the command line, and received the following error: ```Assertion failed: (profile->_multistream.get_profiles().size() > 0), function unsafe_start, file /Users/sanjanasrivastava/langcog-fall/librealsense/src/pipeline/pipeline.cpp, line 84.
Abort trap: 6```. I tried running the permissions script in case that was an issue, it's obviously written for Linux and doesn't work for macOS. The workarounds to run the permissions script seem more invasive than should be necessary, so I don't know if I'm missing something.

I'm not very familiar with C++, so I'm not sure how to trace back from `pipeline.cpp` to `python-tutorial-1-depth.py`. Any help would be much appreciated.

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

Could you try upgrade librealsense to new version?
or run realsense-viewer first?

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@sanjanasrivastava Did you get chance to try with the latest SDK? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@sanjanasrivastava Sorry that we didn't hear from you for weeks. Will close it at this point. Please feel free to create new one if you still have issues or questions. Thanks!

closed 2020-04-08 by RealSenseCustomerSupport