realsenseai/librealsense #4950: can't finish LibUVC install on Raspberry Pi 4

GOBish, 2019-09-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Raspbian buster  | 
| Kernel Version (Linux Only)    |  4.19                                     | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  { }                          | 
| Segment			|                         | 

### Issue Description
I'm trying to install LibUVC on a raspberry pi 4, but when I try
./libuvc_installation.sh
I get to 66% of the process and get the error

[ 60%] Linking CXX shared library librealsense2-gl.so
[ 60%] Built target realsense2-gl
make: *** [Makefile:130: all] Error 2
pi@raspberrypi:~ $ 

I would love some help!

dorodnic, 2019-09-29:
Are you using the script from [libuvc_installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)?

GOBish, 2019-09-29:
https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

dorodnic, 2019-10-02:
I'm waiting to get my Rpi4 to try the installation on. First steps of the script set-up swap file, without which installation fails due to lack of memory. If this is not the reason, I suspect something to do with OpenGL and graphics drivers. For Ubuntu 18 we ask to run:
`sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev`
Probably something similar needs to be done on the Pi.

AnupamYedida, 2019-10-05:
I got it working. 
Just add **set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic")** to the CMakeLists.txt file and compile.

GOBish, 2019-10-05:
@AnupamYedida I really appreciate your help!
Could you walk through the steps for me a little more clearly? I apologize, I don't have a ton of experience with this. How do I get to the cmakelists.txt file?

AnupamYedida, 2019-10-05:
It's in the librealsense-master directory. (Provided you ran the script)

GOBish, 2019-10-06:
Hmm, I still get the same error at 60%.
[ 60%] Linking CXX shared library librealsense2-gl.so
[ 60%] Built target realsense2-gl
make: *** [Makefile:130: all] Error 2

@AnupamYedida are you using Raspbian Buster?

GOBish, 2019-10-06:
I tried it again, making all the changes to cmake files as described here (by @SirDifferential):
 
https://github.com/IntelRealSense/librealsense/issues/4375#issuecomment-509968417

and still could not complete the compile, got this error at 58%

make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/build.make:63: examples/software-de/github.com/IntelRealSense/librealsense/issues/4375#issuecomment-509968417vice/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:511: examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

AnupamYedida, 2019-10-06:
It's the CMakeLists.txt file in the folder **librealsense**

GOBish, 2019-10-06:
Thanks @AnupamYedida -- I did that in https://github.com/IntelRealSense/librealsense/issues/4950#issuecomment-538701444 and was still unsuccesful.
Where in the CMakeLists.txt file did you put that line of code.
Also, I just ran sudo apt-get install libatomic-ops-dev
and I am re-compiling to see if that addresses my issue.

AnupamYedida, 2019-10-06:
I kept it in the top. It fixed the issue for me. I'm running it on **4.19** kernel, Raspberry Pi4, Debian Buster

GOBish, 2019-10-06:
@AnupamYedida I don't know, I can't get it to work no matter what I try.

w3bm0nk3y, 2019-10-06:
This is what worked for me.  Your mileage may vary. 4.19 kernel, Raspberry Pi4, Debian Buster with Neural Computer Stick 2

First run: 
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

Per the LibUVC installation instructions
wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
chmod +x ./libuvc_installation.sh
./libuvc_installation.sh

It will error at:
[ 60%] Linking CXX shared library librealsense2-gl.so
[ 60%] Built target realsense2-gl

find the location of your librealsense_build directory:
pwd ~/librealsense_build

in my case it was /home/pi/librealsense_build

nano libuvc_installation.sh 
or edit with whatever editor you like

In libuvc_installation.sh
change this: 
sudo rm -rf ./librealsense_build
mkdir librealsense_build && cd librealsense_build

to this: 
#sudo rm -rf ./librealsense_build
cd /home/pi/librealsense_build

*replace /home/pi/librealsense_build with your librealsense_build location

Run: sudo ./libuvc_installation.sh

GOBish, 2019-10-07:
@w3bm0nk3y I appreciate the help - but I still get the error at 60% in the compile even following the same steps as you.

GOBish, 2019-10-07:
OK, finally got it! I had to follow all the steps from *both* @AnupamYedida abd @w3bm0nk3y . Thank you all for your help, very much appreciated.

w3bm0nk3y, 2019-10-07:
> OK, finally got it! I had to follow all the steps from _both_ @AnupamYedida and @w3bm0nk3y . Thank you all for your help, very much appreciated.

I realized I forgot to add the steps from @AnupamYedida to my response.  I edited accordingly.

GOBish, 2019-10-07:
Thanks for adding that @w3bm0nk3y ! 
Did you have any issues running the examples after completing the compile? I am having errors in that it can't find the header info when I try to run anything even though I see the folders right there in the correct directory. Thanks!

w3bm0nk3y, 2019-10-08:
Which example are you attempting? I'll try to run it on my side and see if
it works.

GOBish, 2019-10-08:
sorry, added to  my post later -- any of the examples, like rs-enumerate-devices or realsense-viewer. I think there is just some path I have to add somewhere, I just can't figure out where.

soarwing52, 2019-10-08:
I start with a fresh Rasbian on Raspberry pi 4, 4gb RAM
start with [install opencv ](https://www.pyimagesearch.com/2019/09/16/install-opencv-4-on-raspberry-pi-4-and-raspbian-buster/)
and then follow [Raspbian(RaspberryPi3) Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md?source=post_page-----882933271f3d----------------------)

also adding set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -latomic")  to CMAKELIST.txt and it worked with build from source, not the Libuvc

I remember seeing a method putting -D___CXX_FLAGS-latomic can also work, which is giving variable in the terminal, but I can'T find it now

w3bm0nk3y, 2019-10-08:
Right now I am using the python wrappers so I had to go into the
examples/wrappers directory and run everything out of there. You may have
to compile those wrappers as well if they aren't working for you. The pip
install of pyrealsense2 wasn't working for ARM last I checked. Let me know
if that helps and if not I will run the examples and give you screenshots.

GOBish, 2019-10-09:
@soarwing52 - thanks for the input. I tried that, and the compile seems to stop at 57% with the message "Built target rs-benchmark" and no errors - is that what you got, too?

soarwing52, 2019-10-10:
> @soarwing52 - thanks for the input. I tried that, and the compile seems to stop at 57% with the message "Built target rs-benchmark" and no errors - is that what you got, too?

nope, mine stuck at 67% building realsense2-gl, same as in this issue. And it is fixed by adding the cmake -latomic 

And I start with a fresh raspbian, with only pip installed opencv-contrib-python.
I suppose protobuf maybe didn't succeed to build, I used also pip install protobuf(or apt install)

GOBish, 2019-10-10:
@soarwing52 , mine happened with the cmake -latomic added.

dorodnic, 2019-12-10:
Should be resolved now

closed 2019-12-10 by dorodnic

GOBish, 2020-01-12:
@dorodnic I dowloaded the new SDK (downloaded the tar.gz), can you explain to me how I install it on Raspbian now? Thanka

GOBish, 2020-01-12:
Even with the new version, cMake fails at 58% after "built target realsense2-gl"