realsenseai/librealsense #4951: Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors

antoan, 2019-09-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |     4.15.0-64-generic      | 
| Platform                                 | PC  |
| SDK Version                            | 2.29                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I get the above warning after launching realsense viewer, with no devices connected

dorodnic, 2019-09-29:
This warning points out to missing steps during the installation process - in particular:
> 3. Run Intel Realsense permissions script located from librealsense root directory:
./scripts/setup_udev_rules.sh

Alternatively, you could remove the manual install via `sudo make uninstall` and install everything [via apt](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).

sodom89, 2019-09-29:
I get the same error message. After trying to install it with through apt-get the same error occurs.
[image]

sodom89, 2019-09-29:
After uninstalling and reinstalling through the package the error is still occurring. That's why I tried to copy the missing rule (https://raw.githubusercontent.com/IntelRealSense/librealsense/master/config/99-realsense-libusb.rules) manually to the directory /etc/udev/rules.d/99-realsense-libusb.rules. I get a new error now.
[image]

antoan, 2019-10-09:
@dorodni I appreciate that, however running  ./scripts/setup_udev_rules.sh  causes the error message illustrated by @sodom89  last screenshot - "Multiple udev rules found"

sodom89, 2019-10-19:
> This warning points out to missing steps during the installation process - in particular:
> 
> > 1. Run Intel Realsense permissions script located from librealsense root directory:
> >    ./scripts/setup_udev_rules.sh
> 
> Alternatively, you could remove the manual install via `sudo make uninstall` and install everything 
[via apt](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).
Unfortunately, this procedure does not solve the problem. I also reinstalled the system with Ubuntu 18.04 LTS and installed it via the package manager with apt. Unfortunately the error remains.

sodom89, 2019-10-19:
I was able to solve the issue by installing librealsense the manual way.

closed 2019-10-29 by RealSenseCustomerSupport