realsenseai/librealsense #4952: Installation fails when compiling RealSense environment on NVIDIA XT2

liu978079474, 2019-09-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 0.5.11.01.100 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.38+                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | opencv / pcl / python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello,I am running sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u when installing librealsense ,it show Some index files failed to download. They have been ignored, or old ones used instead.

MartyG-RealSense, 2019-09-29:
Have you tried JetsonHacks' guide to installing Librealsense on Jetson TX2 please?

https://www.jetsonhacks.com/2018/07/10/librealsense-update-nvidia-jetson-tx-dev-kits/

liu978079474, 2019-09-29:
> Have you tried JetsonHacks' guide to installing Librealsense on Jetson TX2 please?
> 
> https://www.jetsonhacks.com/2018/07/10/librealsense-update-nvidia-jetson-tx-dev-kits/

Yes，I was tried,but I find the realsense-viewer was 2.13.0.It is very old,and it can not recognize USB 3.0

MartyG-RealSense, 2019-09-29:
There is a method called **libuvc-backend** for attempting Librealsense  installation in a way that does not require kernel patching.  I cannot say if it will work for you, but it does not take long to try it.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2019-09-30:
USB2 indicates that the camera is being detected as being on a USB 2 connection.  This can happen even if the camera is plugged into a USB 3 port.  

Others have had the problem of it being detected as USB2 when using a D435i.  A member of the Intel RealSense team on this forum suggested looking at this link:

https://github.com/IntelRealSense/librealsense/issues/4004

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

Please let us know if you need further assistance. Thank you.

closed 2019-10-24 by RealSenseCustomerSupport