realsenseai/librealsense #4954: is there any way of post-alignment?

dnjfk2003, 2019-09-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |  -                                     | 
| Platform                                 | PC |
| SDK Version                            |   2.19.0                      | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Many of you who use realsense depth camera should know that color and depth image are not aligned if align code is not applied.
Unfortunately, my software which was developed by another person doesn't have alignment code. So color and depth file(png and raw file) is not aligned.
Is there any way make them aligned?

MartyG-RealSense, 2019-09-30:
I hope that the discussion in the link below will be helpful for creating alignment between a color PNG and a depth .raw.

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412810987

dorodnic, 2019-10-02:
You also need calibration (extrinsics and intrinsics) from the camera. Without this information, the answer is no. With it, it would not be easy but can be done. 
We strongly recommend using built-in ROS-bag recorder that captures all necessary data for future post-processing.

95ImadEL, 2019-10-15:
@dorodnic given calibration (extrinsics and intrinsics) from the camera, how can we perform post-processing ? I found that we can do it using software_device, but i'm not sure ?

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

Hi @dnjfk2003,

Did you have a chance to use the align API of RealSense SDK in order to solve your problem?
Please let us know if you need any further help with this matter.

Thank you!

closed 2019-11-08 by RealSenseCustomerSupport