realsenseai/librealsense #4956: How to run on Orange Pi 3

okarpov, 2019-09-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | latest | 
| Operating System & Version |  Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    | 4.9                                        | 
| Platform                                 | Orange Pi 3  |
| SDK Version                            |  2.0 }                          | 
| Language                            |  C++                         | 
| Segment			|  other                          | 

### Issue Description
I would really appreciate any help for building and installing RealSense on Orange Pi 3 ( http://www.orangepi.org/Orange%20Pi%203/ )

Did try with OrangePi_3_xenial_desktop_emmc_v1.1.tar.gz - can build and run RealSense_Viewer but it did not detect camera (even I saw it detected by Ubuntu in log file)

Did try to build but seems something wrong with that on OrangePi_3_xenial_desktop_linux4.9_v1.3.img.tar.gz

please, any suggestion?

MartyG-RealSense, 2019-09-30:
I have only handled one case involving Orange Pi before.  In that case, the camera was detected but it would only work at 480x270 (similar to a Raspberry Pi 3 with USB2).  I will link to that case so that others reading this can get the information.

https://forums.intel.com/s/question/0D50P00004LElHiSAL/fail-to-start-depth-stream-on-a-single-board-pc?language=en_US

Given that Orange Pi is a board where there is only one previous case to refer to, the best course of action may be to try installing Librealsense using a method called l**ibuvc-backend** and see if that works for you.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

okarpov, 2019-10-01:
Thank you for the suggestion, Could you, please, share the build and version of Ubuntu you did install and use?

MartyG-RealSense, 2019-10-01:
libuvc-backend assumes that Ubuntu 16 is being used.

The other Orange Pi user did not say which OS they were using, but their terminal window looks like Ubuntu 18.

dorodnic, 2019-10-02:
I agree that building using the instructions at libuvc_installation.md should be the most advisable for any version of Ubuntu. Let us know if you run into any problems

okarpov, 2019-10-02:
Everything is working as expected and we successfully run realsense_viewer and it now detects camera
Thank you all!

MartyG-RealSense, 2019-10-02:
Great news - thanks for letting us know!  :)

closed 2019-10-15 by dorodnic