realsenseai/librealsense #4957: Fail to capture the depth stream of the 2nd camera with the upgraded FW

szx0112, 2019-09-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 D435i } | 
| Firmware Version                   | (05.11.15) | 
| Operating System & Version |   { Linux (Ubuntu 18.04.3 LTS)  | 
| Kernel Version (Linux Only)    |  (5.0.0-29-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.29.0                      | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<As mentioned. After upgrade the cameras to the new FW, I fail to capture the depth stream of the 2nd camera. Tried both in the realsense-viewer and the Python wrapper. The cameras with old FW not have such issue.>

kjacks21, 2019-10-02:
#4971 is potentially related, at least the portion that I wrote.

kjacks21, 2019-10-03:
I'm using the same specifications as @szx0112 and just to be sure it wasn't the SDK version, I tested SDK v2.23.0 and it could not properly record the depth stream using the latest firmware update.

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @szx0112,

To clarify:
1. Are your both cameras D435i or D435? which firmware version has the trouble? 5.11.15.0?
2. Do you mean you use the realsense-viewer to have the depth streams for both cameras? And, only one of two cameras with the new firmware can have the depth stream normally?

BlackieCen, 2019-11-10:
Hi, I have the same problem as @kjacks21. I select a recorded device from a .bag-file but have the following error.
    depth_sensor = profile.get_device().first_depth_sensor()
RuntimeError: Could not find requested sensor type!
And it worked before well. Using D435 and firmware 05.11.10.0

kjacks21, 2019-11-10:
@BlackieCen try updating your pyrealsense2 version. That fixed into for me.

BlackieCen, 2019-11-10:
@kjacks21 I tried. It works now! Thanks!

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@szx0112, it should be also working for you, right?

szx0112, 2019-11-11:
Yes, the problem is solved, and I am going to close this issue. Thanks.

closed 2019-11-11 by szx0112