realsenseai/librealsense #4958: D435: Getting point cloud becomes slow if camera is moving (C language)

for-no-one, 2019-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version | Ubuntu 18.04.2 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-47-lowlatency  | 
| Platform                                 | PC (Intel® Core™ i3-6100U CPU @ 2.30GHz × 4)  |
| SDK Version                           |  2.29.0-0 | 
| Language                              |  C  | 
| Segment			       |  Robot  | 

### Issue Description
Hello! I'm encountering a very weird issue where my code is working well when camera is stationary, but becomes slow when the camera is moving. Here's my code for extracting a point cloud:

```
rs2_frame_queue* q = rs2_create_frame_queue(1, NULL);

rs2_frame_queue* decimated_queue = rs2_create_frame_queue(1, NULL);
rs2_processing_block* decimation_filter = rs2_create_decimation_filter_block(NULL);

rs2_set_option((rs2_options*)decimation_filter, RS2_OPTION_FILTER_MAGNITUDE, 3, NULL);
rs2_start_processing_queue(decimation_filter, decimated_queue, NULL);

rs2_process_frame(decimation_filter, frame, NULL);

rs2_frame* decimated_frame = rs2_wait_for_frame(decimated_queue, 5000, NULL);

rs2_processing_block* pc = rs2_create_pointcloud(NULL);

rs2_start_processing_queue(pc, q, NULL);

rs2_process_frame(pc, decimated_frame, NULL);

rs2_frame* points = rs2_wait_for_frame(q, 5000, NULL);

rs2_vertex* vtx = rs2_get_frame_vertices(points, NULL);
```

Execution time for `rs2_process_frame(pc, decimated_frame, NULL);` becomes 3-10x slower if I'm shaking the camera. Camera is installed on a mobile robot, so I'm encountering this issue when making sharp turns, but I'm also able to reproduce the issue by shaking the camera by hand. I'm not able to detect the same issue when looking at the 3D image in RealSense Viewer.

ev-mp, 2019-10-02:
@for-no-one , the decimation (post-processing) filter is executed on the CPU, so it is unlikely that it would be directly affected by the camera's vibrations.
The two things that need to be checked:
1) USB Cable - shaking the cable may introduce some degradation of the USB signal quality, resulting in data chunks being re-transmitted and consequently, minor latency.
2) Auto-Exposure - shaking the camera may results the depth camera's auto-exposure algorithms going into oscillations mode with exposure values changing drastically, resulting in FPS drop.
 To mitigate the effect of this you can try to set `exposure priority`  control to `off` (prefer FPS rate over exposure). Ultimately - turn off the depth's sensor `auto-exposure` and  control it manually

for-no-one, 2019-10-02:
@ev-mp Thank you! Turning auto-exposure off fixed the issue. Is there any way to continue using auto-exposure with some modifications (like limiting the maximum exposure time to avoid latency issues) or is manual control the only option? If I set the exposure manually for every frame, does the adjusting the exposure setting add a lot of extra latency?

Also, just for clarification and in case someone else stumbles upon this issue, it was not the decimation filter that was slow, but the point cloud creation. After posting this I tested without the decimation filter and got the same latency issue that seems to happen on the line that converts the image to point cloud. I don't understand why it happens on that line and not during the actual frame grabbing...

for-no-one, 2019-10-02:
@ev-mp Sorry, missed the point about preferring FPS rate over exposure. Will try that next!

for-no-one, 2019-10-02:
@ev-mp Is RS2_OPTION_AUTO_EXPOSURE_PRIORITY the correct option for setting the exposure priority? When I try to set that option to 0 for my depth sensor, I'm getting the following error:
```
rs_error was raised when calling rs2_set_option(options:0000025C32824520, option:Auto Exposure Priority, value:0):
object doesn't support option #30
```

This is my code:
`rs2_set_option((const rs2_options*)sensor, RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0, &e);`

I tried other options like RS2_OPTION_ENABLE_AUTO_EXPOSURE and they worked. Is the exposure priority option not supported by the depth sensor or am I using a wrong option?

ev-mp, 2019-10-02:
@for-no-one , that's correct. Exposure Priority is only available for the RGB sensor.
On the other hand the Depth sensor is predefined to keep constant framerate under all conditions. So exposure settings should not affect the FPS rate.
It could be that if you stream Depth **and** Color streams with identical FPS then you unintentionally invoke RGB->Depth HW synchronization with the Depth stream being slaved to the RGB sensor (that acts as the Master).
But even if that is the case then  the Depth sensor should break the sync and go into Stand-alone mode should the RGB exposure exceed the permitted limit.

Can you confirm that with no post-processing in place you can still observe Depth FPS drop if the camera is vibrating? If that's correct then try to stream Depth only and check whether the issue is reproducible - it will allow to isolate (and possibly address) the issue faster.

for-no-one, 2019-10-02:
@ev-mp How do I prevent streaming from RGB sensor? I don't use RGB data and don't need it. This is what I do now:

```
rs2_frame* frames = rs2_pipeline_wait_for_frames(pipeline, 5000, &e);
int num_of_frames = rs2_embedded_frames_count(frames, &e);
int i;
for (i = 0; i < num_of_frames; ++i)
{
    rs2_frame* frame = rs2_extract_frame(frames, i, &e);
    check_error(e);
    // Check if the given frame can be extended to depth frame interface
    // Accept only depth frames and skip other frames
    if (0 == rs2_is_frame_extendable_to(frame, RS2_EXTENSION_DEPTH_FRAME, &e))
    {
        check_error(e);
        rs2_release_frame(frame);
        continue;
     }
    // post-processing...
}
```

And then I do the post processing. I will test the speed without the post processing in the morning, but if you could help me to make sure that RGB stream is disabled before that, I'd be very grateful.

for-no-one, 2019-10-03:
Looked at the code again, I guess not enabling color stream prevents streaming RGB data... I can confirm that I only enable RS2_STREAM_DEPTH stream. Maybe this is a CPU load issue? Are auto-exposure calculations made on camera on PC hardware? Maybe that's done in a separate thread and is slowing down the post-processing. I will test without post-processing and will also monitor the CPU load next...

for-no-one, 2019-10-03:
@ev-mp If I leave out post-processing and just grab images in a loop, I don't see any significant difference in image grab time between a stationary and a moving camera. So maybe extra CPU load created by auto-exposure calculation happening concurrently with post-processing functions could be the reason for this behavior -- if it's done on the PC. Maybe extreme brightness changes require a lot more calculations compared to normal conditions? I'm able to reproduce the same issue also if I cover the camera with my hand and then uncover it, so variation in brightness seems to cause this.

for-no-one, 2019-10-03:
@ev-mp Thanks, I checked that all my testing was done with only depth stream enabled, so everything discussed above was done with no RGB stream enabled.

I now tried running the C depth example that comes with the library. Without any changes to the example I'm not able to reproduce the issue, but if I add post-processing to the example code, the problem is there. If drastic changes in light condition occur, it sometimes takes over 0.1 seconds to grab the image and process it, while it normally takes less than 0.01 seconds on my system.

Here's my modified rs-depth.c. You should be able to reproduce the issue by covering the lens with your hand.

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

/* Include the librealsense C header files */
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_option.h>
#include <librealsense2/h/rs_frame.h>
#include "../example.h"

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>
#include <time.h>

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                     These parameters are reconfigurable                                        //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define STREAM          RS2_STREAM_DEPTH  // rs2_stream is a types of data provided by RealSense device           //
#define FORMAT          RS2_FORMAT_Z16    // rs2_format is identifies how binary data is encoded within a frame   //
#define WIDTH           640               // Defines the number of columns for each frame or zero for auto resolve//
#define HEIGHT          0                 // Defines the number of lines for each frame or zero for auto resolve  //
#define FPS             30                // Defines the rate of frames per second                                //
#define STREAM_INDEX    0                 // Defines the stream index, used for multiple streams of the same type //
#define HEIGHT_RATIO    20                // Defines the height ratio between the original frame to the new frame //
#define WIDTH_RATIO     10                // Defines the width ratio between the original frame to the new frame  //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

// The number of meters represented by a single depth unit
float get_depth_unit_value(const rs2_device* const dev)
{
    rs2_error* e = 0;
    rs2_sensor_list* sensor_list = rs2_query_sensors(dev, &e);
    check_error(e);

    int num_of_sensors = rs2_get_sensors_count(sensor_list, &e);
    check_error(e);

    float depth_scale = 0;
    int is_depth_sensor_found = 0;
    int i;
    for (i = 0; i < num_of_sensors; ++i)
    {
        rs2_sensor* sensor = rs2_create_sensor(sensor_list, i, &e);
        check_error(e);

        // Check if the given sensor can be extended to depth sensor interface
        is_depth_sensor_found = rs2_is_sensor_extendable_to(sensor, RS2_EXTENSION_DEPTH_SENSOR, &e);
        check_error(e);

        if (1 == is_depth_sensor_found)
        {
            depth_scale = rs2_get_option((const rs2_options*)sensor, RS2_OPTION_DEPTH_UNITS, &e);
            check_error(e);
            rs2_delete_sensor(sensor);
            break;
        }
        rs2_delete_sensor(sensor);
    }
    rs2_delete_sensor_list(sensor_list);

    if (0 == is_depth_sensor_found)
    {
[... 200 lines cut ...]
            loop_count += 1;
        }

        rs2_release_frame(frames);
    }

    // Stop the pipeline streaming
    rs2_pipeline_stop(pipeline, &e);
    check_error(e);

    // Release resources
    free(buffer);
    rs2_delete_pipeline_profile(pipeline_profile);
    rs2_delete_stream_profiles_list(stream_profile_list);
    rs2_delete_stream_profile(stream_profile);
    rs2_delete_config(config);
    rs2_delete_pipeline(pipeline);
    rs2_delete_device(dev);
    rs2_delete_device_list(device_list);
    rs2_delete_context(ctx);

    return EXIT_SUCCESS;
}

```

ev-mp, 2019-10-03:
1. One of the thing that need to be adjusted is to move the creation of processing blocks out of the main loop:
```c++
         rs2_frame_queue* q = rs2_create_frame_queue(1, NULL);

            rs2_frame_queue* decimated_queue = rs2_create_frame_queue(1, NULL);
            rs2_processing_block* decimation_filter = rs2_create_decimation_filter_block(NULL);

            rs2_set_option((rs2_options*)decimation_filter, RS2_OPTION_FILTER_MAGNITUDE, 3, NULL);
            rs2_start_processing_queue(decimation_filter, decimated_queue, NULL);
...
...
            rs2_processing_block* pc = rs2_create_pointcloud(NULL);
            rs2_start_processing_queue(pc, q, NULL);
....
            rs2_delete_processing_block(decimation_filter);
            rs2_delete_frame_queue(decimated_queue);
            rs2_delete_frame_queue(q);
            rs2_delete_processing_block(pc);
```

The processing primitives should be created once in the application's lifetime - failing to do so will have a significant performance impact, and in many cases prevent proper functioning.
Make sure that only the frame's resources are created and released under `while (true)` loop.
2. You need to redefine  the main loop so that `wait_for_frames` will be invoked only once - it will simplify the code and make it much clearer.

As for the variation in processing time - I think that these two factors should be considered:
-Build mode: verify that you build with cmake's flag -DCMAKE_BUILD_TYPE=release
-Number of valid depth points - when you run the `pointcloud` processing block then its performance complexity is O(n). So when you close the lens with hand, you actually cause to generate a depth frame with no/few valid depth points which the filter will gladly accept and process much faster :)
So this is expected.

for-no-one, 2019-10-04:
@ev-mp Thank you! Moving the creation of processing blocks etc out of the loop solved the problem! I didn't realize that creating them every time could prevent proper functioning. I was simply copying commands from post-processing-filters.md and it seemed to work... Maybe it would be worthwhile to mention this in the processing-filters.md or create an example involving post-processing for C language for people lacking common sense like me. :)

Now I only see minor variations in processing time (fully explained by different number of valid points) and I'm down to less than 0.001s processing time compared to occasional 0.1s processing times I was seeing before moving those commands outside the loop.

closed 2019-10-04 by for-no-one