realsenseai/librealsense #4959: Freeze with tkfiledialog

soarwing52, 2019-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 

### Issue Description
Hi, my project is to take road survey images with D435 and gps
and I created a program with tkinter to process, output jpeg, txt, and shapefile

when I import library arcpy, everything works fine.
but if I didn't import arcpy, the tkfiledialog will freeze

´´´
from tkinter import *
from tkinter import filedialog
import pyrealsense2

root = Tk()
file = filedialog.askdirectory()
´´´
this will make the window no response.

I can see that with tkinter is not a common option with rs, or which GUI is tested and would function well?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

This could be similar with #4856. Will further check internally, and see if the community can help this.

soarwing52, 2019-11-01:
another information:
the funciton I use is to create jpg from the bag with multiprocessing.Pool
I can use the tkfiledialog when I import it in the function.

Ex:
```
import tkfiledialog
import multiprocessing as mp

def rs_func():
    import pyrealsense2 as rs

if __name__ =='__main__':
    p = mp.Pool()
    tkfiledialoge()
    p.map()

```

as line 161-217 in my application
[script](https://github.com/soarwing52/RealsensePython/blob/master/data_process.py)

RealSenseCustomerSupport, 2019-11-05:
----------------------------------------------

Hi @soarwing52,

It's good to see your have the workaround to resolve this issue.

closed 2019-11-05 by soarwing52