realsenseai/librealsense #4960: T265 Pipeline start/stop and map start/stop

JBBee, 2019-10-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC |
| SDK Version                            |  2.28.0                          | 
| Language                            |  C++                          | 
| Segment			|  AR                         | 

### Issue Description
Two questions regarding when T265 maps are created in reference to starting and stopping the pipeline:  
- Is a new map, meaning a new position of (0, 0, 0), created each time the pipeline is started?
- When a pipeline is stopped, and the context goes out of scope, does the T265 continue to track its position in its last map?

Thanks.

ev-mp, 2019-10-02:
@JBBee hello,
1) Correct. Each session generates a new origin.
2) To track data stored in previously-recorded maps follow the explanation by @radfordi [here](https://github.com/IntelRealSense/librealsense/issues/4024#issuecomment-496025822), utilizing the  `set_static_node`/`get_static_node` API calls.
#4024

JBBee, 2019-10-02:
Thanks, @ev-mp

closed 2019-10-02 by JBBee