realsenseai/librealsense #4962: Depth Resolution of 848x480 is not allowed anymore

boilerbots, 2019-10-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Linux Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-29-generic   | 
| Platform                                 | PC  |
| SDK Version                            |  v2.29.0 | 
| Language                            |       | 
| Segment			|  Robot  | 

### Issue Description

A couple of problems, first the realsense-viewer is recommending and installing Firmware 5.11.15 but the latest on the website is 5.11.10 (found here https://downloadcenter.intel.com/download/28987/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?product=128255)  and the firmware updater tool document has a link to a page that is either 5.11.11.100 or in the table 5.11.6.250. 

[image]

This doesn't seem to be the source of the problem but I went through several firmware versions trying to debug the problem.

After building and installing the kernel modules on my latest 5.0.0-29 kernel I can no longer select 848x480, I get the following message from ROS but I don't think this is a ROS problem because the same code works on a machine running an older kernel (5.0.0-25-generic).

`[ WARN] [/forward/realsense2_camera_manager]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 848, Height: 480, FPS: 15
`

Further more if I list the support Widths using **lsusb** it isn't there anymore:

**Not working, newer kernel**
`sudo lsusb -d 8086:0b07 -vvv|grep -i Width
        wWidth                            256
        wWidth                            480
        wWidth                            640
        wWidth                           1280
        wWidth                            256
        wWidth                            480
        wWidth                            640
        wWidth                           1280
        wWidth                            256
        wWidth                            480
        wWidth                            640
        wWidth                           1280
        wWidth                            256
        wWidth                            480
        wWidth                            640
        wWidth                           1280
        wWidth                            256
        wWidth                            480
        wWidth                            640
        wWidth                           1280
        wWidth                            424
        wWidth                            640
        wWidth                           1280
FIXME: alloc bigger buffer for device capability descriptors
`

And perhaps the **FIXME** message is an indication of the problem, although I also get that message on the still working but slightly older kernel machine:

**Working computer, older kernel**
`sudo lsusb -d 8086:0b07 -vvv|grep wWidth
        wWidth                            256
        wWidth                            424
        wWidth                            480
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            848
        wWidth                           1280
        wWidth                            256
        wWidth                            424
        wWidth                            480
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            848
        wWidth                           1280
        wWidth                            256
        wWidth                            424
        wWidth                            480
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            848
        wWidth                           1280
        wWidth                            256
        wWidth                            424
        wWidth                            480
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            848
        wWidth                           1280
        wWidth                            256
        wWidth                            424
        wWidth                            480
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            848
        wWidth                           1280
        wWidth                            256
        wWidth                            424
        wWidth                            480
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            848
        wWidth                           1280
        wWidth                           1280
        wWidth                            640
        wWidth                           1280
        wWidth                            320
        wWidth                            320
        wWidth                            424
        wWidth                            640
        wWidth                            640
        wWidth                            848
        wWidth                            960
        wWidth                           1280
FIXME: alloc bigger buffer for device capability descriptors
        wWidth                           1920
        wWidth                           1920
`

ev-mp, 2019-10-02:
@boilerbots hello, 
Can you provide the log generated with the SDK's tool
>rs-enumerate-devices

As well as with the kernel utility
>sudo apt install v4l-utils -y
ls /dev/video* -1a | xargs -n1 v4l2-ctl --list-formats-ext -d

boilerbots, 2019-10-02:
Attached, I have a lot of USB cameras, including 2 D435 devices although only one is attached when I ran the command.

[v4l2-ctl.log](https://github.com/IntelRealSense/librealsense/files/3682021/v4l2-ctl.log)
[rs-enumerate-devices.log](https://github.com/IntelRealSense/librealsense/files/3682022/rs-enumerate-devices.log)

ev-mp, 2019-10-02:
@boilerbots, thank you for the inputs provided.
The reason for not seeing the 848X480 resolution is due to D435 being enumerated only as **USB2** device. An exempt from `rs-enumerate-devices.log`:
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	825312072719
    Firmware Version              : 	05.11.15.00
  **Usb Type Descriptor           : 	2.1**

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Depth	  1280x720	@ 6Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  640x480	@ 15Hz	   Z16
    Depth	  640x480	@ 6Hz	   Z16
    Depth	  480x270	@ 60Hz	   Z16
    Depth	  480x270	@ 30Hz	   Z16
    Depth	  480x270	@ 15Hz	   Z16
    Depth	  480x270	@ 6Hz	   Z16
    Depth	  256x144	@ 90Hz	   Z16

USB2 configuration supports only a subset of depth streaming profiles, as depicted above.
If the device was able to stream 848X480 on the test PC, then the change is most certainly due to the cable/USB port issue. Please check the cabling and try to replug/switch port until you find a stable USB3 configuration.

boilerbots, 2019-10-02:
OMG I can't believe it, our tech switch out the cables in our system and they are now USB2 cables with a type C plug on the end. Every time I tested the "new" kernel I plugged my laptop into that system using those cables. Every time I tested the "old" kernel using my desktop machine I had to use a longer and proper USB3 cable.

So sorry.

closed 2019-10-02 by boilerbots