realsenseai/librealsense #4963: Need to Stabilize Frame Rates for D-435 in C# (disable Auto Exposure Priority?)

brenteickhoff, 2019-10-01:
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 series 
| Firmware Version                   |  05.11.06.250
| Operating System & Version |  Windows 10      
| Platform                                 |  PC
| SDK Version                           |  2.19.1                   
| Language                               | C#
| Segment			         | Healthcare Research, AI Deep Learning, Motion Pattern Recognition               

### Issue Description

- I would like to try to stabilize FPS (frame rate) under various lighting conditions.
- My application is currently based on the C# Wrapper Sample #2 (with the addition of saving to ROS.bag)

- Not getting 30 fps @ 640x360 under office lighting (getting 22fps). 
- Dropped resolution to 424x240 helped but now 29fps and still not constant. 
- Dropped to 15fps and getting 16fps

- Is there a way to stabilize frame rate?
- How can I disable Auto Exposure Priority in C#?
- What else can I do to stabilize frame rate? It seems inconsistent.

Thank you in advance,
Brent

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

The auto exposure priority function can be disabled by something like this, so the fps would not drop under low light condition:
`colorSensor.Options[Option.AutoExposurePriority].Value = 0f;`
And the auto exposure can be disabled by:
`colorSensor.Options[Option.EnableAutoExposure].Value = 0f;`

Please let me know if there is any further question. Thank you.

brenteickhoff, 2019-10-23:
Thank you for your reply.
My application is based on C# Wrapper Sample 2, I don't have any Sensor objects.
I have only Pipeline, PipelineProfile, Colorizer, and Config objects. Config.EnableStream, Pipeline.WaitForFrames.
Would I need to take another approach?
Thanks, Brent

RealSenseCustomerSupport, 2019-10-28:
----------------------------------------------

You can get Sensor object from PipelineProfile
Here is an example for getting the depth sensor:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#get-depth-units
Thank you.

closed 2019-11-01 by RealSenseCustomerSupport