realsenseai/librealsense #4964: about Intel RealSence D435i SDK

yoichi0110, 2019-10-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (0.5.11.06.250) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.25.0.993 }                          | 
| Language                            |  {C#/C++ }                          | 
| Segment			|  {PC }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

An exception occurs in pipe-> start () of the following program. The exception is said to be a camera that is already streaming. Why?

void rs2start(const char *str)
{
	try{
		// Declare frameset and frames which will hold the data from the camera
		rs2::frameset frames;
		rs2::frame depth;

		// Declare depth colorizer for pretty visualization of depth data
		rs2::colorizer color_map;

		// Create a shared pointer to a pipeline
		auto pipe = std::make_shared<rs2::pipeline>();

		// Start streaming with default configuration
		pipe->start();

		// Initialize a shared pointer to a device with the current device on the pipeline
		rs2::device device = pipe->get_active_profile().get_device();

		// If the device is sreaming live and not from a file
		if (!device.as<rs2::playback>())
		{
			frames = pipe->wait_for_frames(); // wait for next set of frames from the camera
			depth = color_map.process(frames.get_depth_frame()); // Find and colorize the depth data
		}

		if (!device.as<rs2::playback>()) // Disable recording while device is playing
		{
			if (!device.as<rs2::recorder>())
			{
				pipe->stop(); // Stop the pipeline with the default configuration
				pipe = std::make_shared<rs2::pipeline>();
				rs2::config cfg; // Declare a new configuration
				cfg.enable_record_to_file(str);
				pipe->start(cfg); //File will be opened at this point
			}
		}
	}
	catch (const rs2::error & e)
	{
		std::cout << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
	}
	catch (const std::exception& e)
	{
		std::cerr << e.what() << std::endl;
	}
}

ev-mp, 2019-10-02:
@yoichi0110 , the function you posted called `void rs2start(const char *str)` and it represents only part of the job, also you do not provide the flow - is this function called once or multiple times ?
There presumably should be a symmetric `void rs2stop()` which should implement `pipe->stop()` and which must be invoked to make the camera `open/close` calls consistent.

You can follow the SDK example that you are reusing (looks similar to `rs-record-playback`) to see the full `pipe` utilization flow.

yoichi0110, 2019-10-02:
void rs2start (const char * str) is read only once. Also, make void rs2stop () read only once. However, rs2start raises an exception that points out that the camera is already streaming, and rs2stop raises an exception that states that stop () cannot be called before start (). I'm at a loss without knowing why. Please give me a professor.

ev-mp, 2019-10-03:
@yoichi0110 , sorry, but again, there are gaps in data and flow for a proper analysis. 
Several notes:
>// Create a shared pointer to a pipeline
   auto pipe = std::make_shared<rs2::pipeline>();

The `pipeline`is defined only in the local scope. See below

>rs2::device device = pipe->get_active_profile().get_device();

Unused code
>if (!device.as<rs2::playback>())
{
	frames = pipe->wait_for_frames(); // wait for next set of frames from the camera
	depth = color_map.process(frames.get_depth_frame()); 

You should to check that `frames` object actually includes a `depth` frame to avoid nullptr de-referencing.
In the following section you restart streaming with recording enabled:
```c++
if (!device.as<rs2::playback>()) // Disable recording while device is playing
{
	if (!device.as<rs2::recorder>())
	{
		pipe->stop(); // Stop the pipeline with the default configuration
		pipe = std::make_shared<rs2::pipeline>();
		rs2::config cfg; // Declare a new configuration
		cfg.enable_record_to_file(str);
		pipe->start(cfg); //File will be opened at this point
	}
}
```

What is the purpose of destroying a just-created `pipeline` and building a new one? The code is fully deterministic so the first pipe is redundant here, not mentioning the time overhead involved.

2. When you say that you close the pipe from another function named `rs2stop()`  - I see that the `pipe` object is defined only locally:
>// Create a shared pointer to a pipeline
   auto pipe = std::make_shared<rs2::pipeline>();

so it is not clear whether you manage to pass that `pipe` to the function `rs2stop()` in order to call `pipe->stop()`.

Altogether the function's code is not well-formed. It would be useful to provide an explanation of the actual use-case. Then we would be able to refer you to the appropriate code/sample to follow.

yoichi0110, 2019-10-03:
The fact that the rs2start () and rs2stop cannot be used properly is understood from the fact that the pipeline is defined only in the local scope. I don't know the reason for the exception being generated. Professor

void rs2start (const char * str) try
{
auto pipe = std :: make_shared <rs2 :: pipeline> ();
rs2 :: config cfg; // Declare a new configuration
cfg.enable_record_to_file (str);
pipe-> start (cfg); // File will be opened at this point
}

yoichi0110, 2019-10-03:
The fact that the rs2start () and rs2stop cannot be used properly is understood from the fact that the pipeline is defined only in the local scope. I don't know the reason for the exception being generated. Professor

void rs2start (const char * str) try
{
auto pipe = std :: make_shared <rs2 :: pipeline> ();
rs2 :: config cfg; // Declare a new configuration
cfg.enable_record_to_file (str);
pipe-> start (cfg); // File will be opened at this point
}
catch (const rs2 :: error & e)
{
std :: cout << "RealSense error calling" << e.get_failed_function () << "(" << e.get_failed_args () << "): \ n" << e.what () << std :: endl;
}
catch (const std :: exception & e)
{
std :: cerr << e.what () << std :: endl;
}

ev-mp, 2019-10-03:
@yoichi0110 , what is the full error you receive ? Can you verify that the camera is properly recognized with SDK's tools/demos?

yoichi0110, 2019-10-03:
The SDK works fine with record-playback, but an exception occurs in the project I created.

ev-mp, 2019-10-03:
What is the actual exception message?
Could it be that the file you're trying to write to is locked/permission-denied ?

yoichi0110, 2019-10-03:
The actual exception message is caught by the catch statement as in the above program, and the message is as follows. "RealSense error calling rs2_pipeline_start_with_config (pipe: 0000029EF2735080, config: 0000029EF2734E80):
     Camera already streaming "
Then, 5KB is generated for the bag file, and it seems that the file is not locked.
The rs2start function this time is called from C #, so it is made into a DLL. Is this affecting this error?

ev-mp, 2019-10-03:
There is nothing in the snippet that would suggest that the pipeline is activated twice, unless `pipe-start()` is also called elsewhere, or the `pipe-stop()` is not invoked between consecutive calls.
Debug the calls to `pipe->start()` to verify the code flow is properly configured. Note that you have a working reference to compare to.

Another less-likely explanation is that the camera entered some unresponsive state. This can be easily verified by running SDK's demos. And if you find that none of SDK original demos can stream the camera then re plugging the device should fix it.

yoichi0110, 2019-10-03:
pipe-> start is executed only once. Pipe-> stop is not running either. For reference, if you execute pipe-> stop instead of pipe-> start, the following exception is output. ”Stop () cannot be called before start ()”
Also, if you run the SDK record-playback, it will work normally, so it is not a situation where the camera does not respond. If you run pipe-> start for the project I created, the camera is already streaming.

ev-mp, 2019-10-10:
@yoichi0110 , building the code with `TRACE_API=true`  flag would allow to monitor all the SDK calls invoked from the user code.
Then you can post the full log generated in the reproduction scenario for review

yoichi0110, 2019-10-10:
What should I do to see the logs?

ev-mp, 2019-10-10:
The native log APIs  are ['log_to_console(' and `log_to_file(..`](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/NativeMethods.cs#L547-L551).
THose are wrapped for convenience in [`Log` class](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/Helpers/Log.cs)

yoichi0110, 2019-10-11:
where is 'log_to_console(' and log_to_file(..

yoichi0110, 2019-10-11:
Is it the following function?
void log_to_console (rs2_log_severity min_severity)
         {
             minimum_console_severity = min_severity;
             open ();
         }

void log_to_file (rs2_log_severity min_severity, const char * file_path)
         {
             if (! try_get_log_severity (minimum_file_severity))
                 minimum_file_severity = min_severity;

             if (file_path)
                 filename = file_path;

             open ();
         }

Looking at this function, it was found that a log file was generated. However, the log file output destination could not be found.

yoichi0110, 2019-10-17:
The exception no longer occurs when the caller is changed.
Even if the following program is executed, it is not recorded. Why?

std :: shared_ptr <rs2 :: pipeline> strPipe;
void rs2start (const char * str) try
{
　　Std :: make_shared <rs2 :: pipeline> ();
　　auto pipe = std :: make_shared <rs2 :: pipeline> ();
　　strPipe = pipe;

　　rs2 :: config cfg; // Declare a new configuration
　　cfg.enable_record_to_file (str);
　　pipe-> start (cfg); // File will be opened at this point
}
catch (const rs2 :: error & e)
{
　　std :: cout << "RealSense error calling" << e.get_failed_function () << "(" << e.get_failed_args () << "): \ n" << e.what () << std :: endl;
}
catch (const std :: exception & e)
{
　　std :: cerr << e.what () << std :: endl;
}

yoichi0110, 2019-10-18:
When pipe-> start is executed from the main function, it ends normally, but after executing Application.Run, an exception occurs at pipe-> start. Why?

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@yoichi0110 What's the exception details when you execute "Application.Run"? Any logs or snapshots? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@yoichi0110  Could you please provide more about the exception details so that we can further look into it? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@yoichi0110 We didn't get response from you for weeks, will close this issue at this point. Please feel free to create new one if you still have issues or questions. Thanks!

closed 2020-02-25 by RealSenseCustomerSupport