realsenseai/librealsense #4965: ModuleNotFoundError: No module named 'pybackend2'

Dontla, 2019-10-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.25.0      | 
| Language                            | python }                          | 
| Segment			|  Robot                   | 

### Issue Description

When I ran the example at [https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/pybackend_example_1_general.py](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/pybackend_example_1_general.py), it occured the error:
ModuleNotFoundError: No module named 'pybackend2'
, import pyrealsense2 was ok, but the 'pybackend2' was not, why?
[image]
[image]
[image]

MartyG-RealSense, 2019-10-02:
I hope that the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/3204#issuecomment-460276194

MartyG-RealSense, 2019-10-02:
I see that you found the advice about pybackend2.   Does this advice not work for you, please?

https://github.com/IntelRealSense/librealsense/issues/3204#issuecomment-460276194

Dontla, 2019-10-02:
No, it not work, give me more assistance please~

MartyG-RealSense, 2019-10-02:
I looked carefully at the example Python program you are using.  It seems to be similar to the **rs-enumerate-devices** C++ program, in that it lists the details of attached devices and offers some other commands too.

An alternative to using the pybackend sample to list the devices may be the script in th elink below offered by a RealSense community member.

https://github.com/IntelRealSense/librealsense/issues/2332#issuecomment-419959696

dorodnic, 2019-10-02:
Everything you need to do with a RealSense device can be accomplished with pyrealsense2.
Unless there is very specific reason to use it, I would recommend to ignore pybackend and its sample.

RealSenseCustomerSupport, 2019-10-21:
----------------------------------------------

Hi @Dontla,

Do you need any further help with this question?

Thank you!

closed 2019-10-28 by RealSenseCustomerSupport