realsenseai/librealsense #4968: Which function in python wrapper can I use to operate specific camera with its serial number?

Dontla, 2019-10-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435&D435i| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python                           | 
| Segment			|  Robot                       | 

### Issue Description
I use six or more cams in one system, I want to operate specific camera with its serial number, however, I can't find function in the [python wrapper](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python)...could you help me

dorodnic, 2019-10-02:
There should be a version of `enable_stream` on config class that accepts serial number string as parameter. From there you pass the config object to `pipeline.start`

Dontla, 2019-10-05:
I have read the doc in config.py, but I still can't know how to implement it... are there any code samples...

RealSenseCustomerSupport, 2019-10-28:
----------------------------------------------

Hi @Dontla,

Try below example code to enable the specific camera:

    pipeline = rs.pipeline()
    config = rs.config()

    config.enable_device(ds5_serial)
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

     # Start streaming
    pipeline.start(config)

Dontla, 2019-11-25:
I have tried it! it worked! We can use context object to get our camera's serials number, then use config object to enable it!
`ctx = rs.context()
serial_0 = ctx.devices[0].get_info(rs.camera_info.serial_number)
serial_1 = ctx.devices[1].get_info(rs.camera_info.serial_number)
config_0.enable_device(serial_0)
config_1.enable_device(serial_1)
`
[image]
Thank you!

closed 2019-11-29 by RealSenseCustomerSupport