realsenseai/librealsense #4969: Jetson TX2 installation in the october of 2019

jungladicitta, 2019-10-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435, T265 } | 
| Firmware Version                   | latest | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  (4.4.38)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  not installed yet              | 
| Segment			|  Robot                           | 

### Issue Description

To install librealsense on Jetson TX2 I have read these readme files: 
[installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
[distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)
[JetsonHacks: buildLibrealsense2TX/README.md](https://github.com/jetsonhacks/buildLibrealsense2TX/blob/master/README.md)

So, as I understood to install librealsense on Jetson TX2 I should follow instructions from distribution_linux.md except for one moment about kernel patching. 
Here is the problem.
Instruction from installation_jetson.md is inapplicable because instructions contains out-of-date links to patch scripts.
Another way to patch kernel is to run JetsonHacks scripts.
buildPatchedKernel.sh from JetsonHacks is inapplicable because I dont have kernel in my /usr/src/kernel directory. Moreover, installLibrealsense.sh installs librealsense v2.13.0 instead of latest 2.25.
Do I understand something wrong?
What can I do to patch kernel and install librealsense properly on my Jetson TX2?

dorodnic, 2019-10-03:
We will update installation_jetson.md soon with new instructions. 

> distribution_linux.md except for one moment about kernel patching

This is correct. 

> What can I do to patch kernel and install librealsense properly on my Jetson TX2?

We no longer recommend kernel patching for Jetson installation. Installing the SDK via apt should be entirely sufficient.

brandonhanner, 2019-10-11:
@dorodnic  is there any time frame on these updates? Installing via apt seems to not work at the moment (apt complains that there is no release for arm64).

studennis911988, 2019-10-15:
@dorodnic Really looking forward it!  Is there any expected time?

dorodnic, 2019-10-15:
@studennis911988 - I just created a pull-request with updated instructions. 
@brandonhanner - can you share more details? Are you using Ubuntu 16 or 18? (I think we published the 18 version also but I may be mistaken) Did you get any errors during the process?

studennis911988, 2019-10-15:
@dorodnic  
First of all, really appreciate all your hard-work.
Before jump right into the new installation, I have two concerns.
1. Does it mean that the new installation process is independent to the Jetpack or L4T version?
2. Does the new installation tested on D435i? Since I have noticed that **_only firmware 5.10.03 is able to connect D435i to my NVIDIA TX2 , moreover, D435i can only be recognized as D435 and the emitter is disable in this firmware_**. [Just like this issue](https://github.com/IntelRealSense/realsense-ros/issues/835#issuecomment-524582529)

sjdrc, 2019-10-16:
Does the nvidia provided kernel now have support for frame metadata?

jungladicitta, 2019-10-25:
@dorodnic thanks for your work and reply!

I have installed librealsense via apt on my jetson tx2 and I got d435 working in ros, but after some time it stops working and I see this error:
```
[camera/realsense2_camera_manager-5] process has died [pid 3038, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nvidia/.ros/log/6312e40e-f6fb-11e9-9e19-b42e994ddade/camera-realsense2_camera_manager-5.log].
log file: /home/nvidia/.ros/log/6312e40e-f6fb-11e9-9e19-b42e994ddade/camera-realsense2_camera_manager-5*.log
```
[image]

What does mean `code -11`? Is this related to wrong librealsense installation on jetson tx2?

studennis911988, 2019-10-26:
@jungladicitta  @dorodnic 
I had installed SDK successfully and the performance is quite nice, but I encountered the same issue as @jungladicitta does, the ros-wrapper dumped occasionally and the **_system become very lag_** when using ros-wrapper even if the cpu use rate is under 50%.
@dorodnic wish you can give as some hint, thanks in advanced!

jungladicitta, 2019-10-30:
@dorodnic 
Also I have issue that t265 disconnects when d435 is connected even with using powered USB 3.0 hub. Check out the description https://github.com/IntelRealSense/librealsense/issues/5129
What do you think?

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

The original issue (librealsense installation on TX2) should already be addressed.
Please feel free to submit a another ticket if more assistance is needed.
Closing this ticket unless otherwise. Thank you.

closed 2020-02-11 by RealSenseCustomerSupport