realsenseai/librealsense #4970: Error installing Ubuntu 4.16 patch

evercoast-smm, 2019-10-02:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.16.18-041618-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

Running `.../librealsense-2.29.0$ ./scripts/patch-ubuntu-kernel-4.16.sh` fails to complete, see:

```
.../librealsense-2.29.0$ ./scripts/patch-ubuntu-kernel-4.16.sh
Development script for kernel 4.16 with metadata node
[sudo] password for evercoast: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
build-essential set to manually installed.
git is already the newest version (1:2.17.1-1ubuntu0.4).
linux-headers-generic is already the newest version (4.15.0.65.67).
linux-headers-generic set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Package required bison:  - not found, installing now...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libbison-dev
Suggested packages:
  bison-doc
The following NEW packages will be installed:
  bison libbison-dev
0 upgraded, 2 newly installed, 0 to remove and 3 not upgraded.
Need to get 605 kB of archives.
After this operation, 1,811 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libbison-dev amd64 2:3.0.4.dfsg-1build1 [339 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 bison amd64 2:3.0.4.dfsg-1build1 [266 kB]
Fetched 605 kB in 1s (528 kB/s) 
Selecting previously unselected package libbison-dev:amd64.
(Reading database ... 286721 files and directories currently installed.)
Preparing to unpack .../libbison-dev_2%3a3.0.4.dfsg-1build1_amd64.deb ...
Unpacking libbison-dev:amd64 (2:3.0.4.dfsg-1build1) ...
Selecting previously unselected package bison.
Preparing to unpack .../bison_2%3a3.0.4.dfsg-1build1_amd64.deb ...
Unpacking bison (2:3.0.4.dfsg-1build1) ...
Setting up libbison-dev:amd64 (2:3.0.4.dfsg-1build1) ...
Setting up bison (2:3.0.4.dfsg-1build1) ...
update-alternatives: using /usr/bin/bison.yacc to provide /usr/bin/yacc (yacc) in auto mode
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Missing package installed
Package required flex:  - not found, installing now...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libfl-dev libfl2
Suggested packages:
  flex-doc
The following NEW packages will be installed:
  flex libfl-dev libfl2
0 upgraded, 3 newly installed, 0 to remove and 3 not upgraded.
Need to get 334 kB of archives.
After this operation, 1,113 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 flex amd64 2.6.4-6 [316 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libfl2 amd64 2.6.4-6 [11.4 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libfl-dev amd64 2.6.4-6 [6,320 B]
Fetched 334 kB in 0s (739 kB/s)      
Selecting previously unselected package flex.
(Reading database ... 286773 files and directories currently installed.)
Preparing to unpack .../flex_2.6.4-6_amd64.deb ...
Unpacking flex (2.6.4-6) ...
Selecting previously unselected package libfl2:amd64.
Preparing to unpack .../libfl2_2.6.4-6_amd64.deb ...
Unpacking libfl2:amd64 (2.6.4-6) ...
Selecting previously unselected package libfl-dev:amd64.
Preparing to unpack .../libfl-dev_2.6.4-6_amd64.deb ...
Unpacking libfl-dev:amd64 (2.6.4-6) ...
Setting up flex (2.6.4-6) ...
Setting up libfl2:amd64 (2.6.4-6) ...
Setting up libfl-dev:amd64 (2.6.4-6) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for install-info (6.5.0.dfsg.1-2) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Missing package installed
Package required libelf-dev:  - not found, installing now...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libelf-dev
0 upgraded, 1 newly installed, 0 to remove and 3 not upgraded.
Need to get 57.3 kB of archives.
After this operation, 374 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libelf-dev amd64 0.170-0.4ubuntu0.1 [57.3 kB]
Fetched 57.3 kB in 0s (221 kB/s)    
Selecting previously unselected package libelf-dev:amd64.
(Reading database ... 286863 files and directories currently installed.)
Preparing to unpack .../libelf-dev_0.170-0.4ubuntu0.1_amd64.deb ...
Unpacking libelf-dev:amd64 (0.170-0.4ubuntu0.1) ...
Setting up libelf-dev:amd64 (0.170-0.4ubuntu0.1) ...
Missing package installed
Package required libelf-dev:  - found
Package required elfutils:  - not found, installing now...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libasm1
The following NEW packages will be installed:
  elfutils libasm1
0 upgraded, 2 newly installed, 0 to remove and 3 not upgraded.
Need to get 262 kB of archives.
After this operation, 931 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 libasm1 amd64 0.170-0.4ubuntu0.1 [14.4 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu bionic-updates/main amd64 elfutils amd64 0.170-0.4ubuntu0.1 [247 kB]
Fetched 262 kB in 0s (650 kB/s)   
Selecting previously unselected package libasm1:amd64.
(Reading database ... 286875 files and directories currently installed.)
Preparing to unpack .../libasm1_0.170-0.4ubuntu0.1_amd64.deb ...
Unpacking libasm1:amd64 (0.170-0.4ubuntu0.1) ...
Selecting previously unselected package elfutils.
Preparing to unpack .../elfutils_0.170-0.4ubuntu0.1_amd64.deb ...
Unpacking elfutils (0.170-0.4ubuntu0.1) ...
Setting up libasm1:amd64 (0.170-0.4ubuntu0.1) ...
Setting up elfutils (0.170-0.4ubuntu0.1) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Missing package installed
Cloning into './ubuntu-bionic-v4.16.18'...
remote: Counting objects: 66923, done.
remote: Compressing objects: 100% (63588/63588), done.
remote: Total 66923 (delta 7699), reused 25026 (delta 2348)
Receiving objects: 100% (66923/66923), 175.19 MiB | 1.73 MiB/s, done.
Resolving deltas: 100% (7699/7699), done.
Note: checking out '62e9ccfaaedffd057e921fca976f9f7f71c9b254'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Checking out files: 100% (62930/62930), done.
Applying realsense-uvc patch
./scripts/patch-ubuntu-kernel-4.16.sh: line 106: ../scripts/realsense-camera-formats_ubuntu-bionic-v4.16.patch: No such file or directory
```

ev-mp, 2019-10-02:
@evercoast-smm hello,
The kernel patch for 4.16 was developed with Xenial distribution:
>./scripts/patch-ubuntu-kernel-4.16.sh: line 106: ../scripts/realsense-camera-formats_ubuntu-**bionic**-v4.16.patch: **No such file or directory**

In general only Ubuntu LTS kernel patches are maintained. (4.4, 4.15, 4.18..). Is there any particular reason to use non-LTS kernel with Ubuntu 18?

evercoast-smm, 2019-10-04:
Thank you. We've moved onto 4.18.20-041820-generic.

closed 2019-10-04 by evercoast-smm