realsenseai/librealsense #4972: Freeze on exit for Realsense Viewer and most rs-* example apps

evercoast-smm, 2019-10-02:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.16.18-041618-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

Issue:

We're seeing Realsense Viewer and all of the rs-* example apps freeze on exit (notably, rs-enumerate-devices is OK).

To reproduce:

1) Start `realsense-viewer`, `rs-capture`, or etc...

2) Press the close (X) button in the upper right of the window

3) The application will have frozen

Our primary questions is:

- Is this behavior normal, i.e. a known bug? Otherwise, might this be indicative of a problematic machine configuration. And if so, how can this be rectified?

Additional info:

Please note that we've also filed this issue, which could be related: https://github.com/IntelRealSense/librealsense/issues/4970

ev-mp, 2019-10-02:
@evercoast-smm hello, please check whether the command-line demos, such as `rs-depth` run smoothly.
If that's correct then the video subsystem works properly, and the issue is elsewhere, probably with the graphic drivers.

evercoast-smm, 2019-10-10:
@ev-mp, thank you for your help. First, we've moved to kernel `4.18.20-041820-generic` as suggested to align to LTS.

Next, running `rs-depth` we see the following:

```
evercoast@evercoast:~/SrcEvercoast/Mavericks2_API$ rs-depth
There are 5 connected RealSense devices.

Using device 0, an Intel RealSense D415
    Serial number: 833412060081
    Firmware version: 05.11.15.00

[followed by thousands of blank lines here]                                                            
```

Subsequently, for comparison, running `rs-color` we see the following:

```
evercoast@evercoast:~/SrcEvercoast/build_MavericksDependencies/librealsense-2.29.0/include$ rs-color
There are 5 connected RealSense devices.

Using device 0, an Intel RealSense D415
    Serial number: 833412060081
    Firmware version: 05.11.15.00

RGB frame arrived.
First 10 bytes: 15 18 17 15 18 17 14 1a 18 13 
Frame No: 1
Timestamp: 1570715579548.069092
Timestamp domain: Global Time
Time of arrival: 1570715579580

RGB frame arrived.
First 10 bytes: 12 18 18 13 19 19 15 18 19 14 
Frame No: 2
Timestamp: 1570715579581.685791
Timestamp domain: Global Time
Time of arrival: 1570715579614

RGB frame arrived.
First 10 bytes: 0f 1a 16 0f 1a 16 10 1b 19 0f 
Frame No: 3
Timestamp: 1570715579615.293701
Timestamp domain: Global Time
Time of arrival: 1570715579647
```

Also, running `rs-capture` yields the following errors, and again crashes on exit:
```
evercoast@evercoast:~/SrcEvercoast/build_MavericksDependencies/librealsense-2.29.0/include$ rs-capture 
 10/10 09:53:23,247 ERROR [140569807902592] (backend-v4l2.cpp:555) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
unique_id- 36-4-4
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:01.0/0000:9b:00.0/usb36/36-4/36-4:1.3/video4linux/video28
susb specification- 320
metadata node-/dev/video29 was already been assigned 
 10/10 09:53:23,614 ERROR [140569807902592] (backend-v4l2.cpp:555) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
unique_id- 36-4-4
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:01.0/0000:9b:00.0/usb36/36-4/36-4:1.3/video4linux/video28
susb specification- 320
metadata node-/dev/video29 was already been assigned 
 10/10 09:53:23,624 ERROR [140569807902592] (backend-v4l2.cpp:555) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
unique_id- 36-4-4
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:01.0/0000:9b:00.0/usb36/36-4/36-4:1.3/video4linux/video28
susb specification- 320
metadata node-/dev/video29 was already been assigned 
 10/10 09:53:23,631 ERROR [140569807902592] (backend-v4l2.cpp:555) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
unique_id- 36-4-4
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:01.0/0000:9b:00.0/usb36/36-4/36-4:1.3/video4linux/video28
susb specification- 320
metadata node-/dev/video29 was already been assigned 
 10/10 09:53:23,639 ERROR [140569807902592] (backend-v4l2.cpp:555) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- ad3
mi- 3
unique_id- 36-4-4
path- /sys/devices/pci0000:98/0000:98:00.0/0000:99:00.0/0000:9a:01.0/0000:9b:00.0/usb36/36-4/36-4:1.3/video4linux/video28
susb specification- 320
metadata node-/dev/video29 was already been assigned 

#### RS Frame Rate Printer ####
09:53:23.897 [17714] [E] infra/Poller: poll: epoll_wait error 4
09:53:23.897 [17714] [E] Dispatcher: handleEvents(): Poller::poll() ret -1
```

Thank you!

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @evercoast-smm,

Please run realsense-viewer and rs-capture having only one camera connected. Let us know
whether or not any of them freezes on exit.

Thank you!

closed 2019-11-12 by RealSenseCustomerSupport