realsenseai/librealsense #4974: How long will the d435 be available?

TyTruGolf, 2019-10-03:
We're making a case designed to hold the d435 camera and the question that came up is will this camera still be for sale in one or two years or will it be phased out once a newer model becomes available?

We prefer the camera to the board since we're a small company and getting the camera prebuilt helps to reduce manufacturing issues, but would we be better off long term building a design around the d430 module?

Thanks, and sorry if this is not the right place to ask the question.

MartyG-RealSense, 2019-10-03:
It is a difficult question.  As I am not an Intel employee, the comments below should not be taken as an official position.

The SR300 was briefly retired and then made a comeback when RealSense SDK 2.0 launched.  It was eventually no longer referred to as a retired camera, and was then relaunched as the SR305, the new entry-level model of the RealSense family.

Likewise, the 400 Series was projected in a public roadmap shown at the end of 2017 to have a 3 year life cycle til end 2019 (the longest RealSense life cycle yet) and then a new RealSense generation would be introduced.  Three years is forever in technology and plans have undoubtedly changed, though new RealSense models are currently undergoing testing - Intel are always working on RealSense projects, though not every project ends up being released.

There is no public information that can currently be given about forthcoming models.  You may be interested though in an upcoming joint webinar between Intel and FRAMOS on October 10, where the RealSense portfolio will be discussed as part of that event.

https://github.com/IntelRealSense/librealsense/issues/4944

As the next generation of models are likely to use some form of the current SDK software, there is likely to be overlap between generations in the same way that the SR300 / SR305 and the 400 Series currently share the same SDK space.

As FRAMOS have just launched their own D435e own-brand camera aimed at industry, there is likely to be life in the 400 Series technology for some time to come.

agrunnet, 2019-10-03:
My two cents:
We have a typical lifecycle of three years but do not set hard EOL’s for our cameras.  For example we expect our D400 family to be a viable choice well through 2021 (4 years after launch).  We do focus on making our software forward and backward compatible so if we do EOL after 3 years all the work a customer done is not lost with a new family launch

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

@TyTruGolf Do you have further question about this topic?

closed 2019-11-01 by RealSenseCustomerSupport