realsenseai/librealsense #4977: When will Matlab wrappers in OSX become available?

SteveDSP, 2019-10-04:
Camera Model                      D400 
Operating System & Version  MacOS  : 10:14:4
Platform                               Mac   / Xcode
SDK Version                         2.29
Language                             C++ / Matlab                       |

This is a feature request / bug fix

If you try to build the Matlab wrapper functions on Xcode, the compilation fails with errors in factory.h and .cpp files. As I understand things as they stand, only Windows is supported thus far to enable Matlab wrappers.

1. Is there a work around that anyone else has tried to enable the Matlab wrappers to compile in Xcode and therefore enable RealSense use in Matlab under OSX?

2. If there is no workaround, when is the support for this going to be available please? i.e bug sorted etc.

On the same same subject, there are mixed messages as far as the whole install/compilation for OSX environment. The intel website OSX install differs from that on the GitHub site itself. It would be wise to update the current Intel website ones with the Github version (the one that I got to work at least) ..albeit with a further vulkan-sdk fix needed when doing the homebrew install.

brew cask install vulkan/vulkan-sdk ...results in vulkan-sdk not found
brew cask install apenngrace/vulkan/vulkan-sdk ...works (but only as long as Mr Grace exists)

And..there may still be issues with that (see https://vulkan.lunarg.com/issue/home?limit=10;q=;mine=false;org=false;khronos=false;lunarg=false;indie=false;status=new,open)

The best way to install Vulcan is here (if it is a requirement)..

https://vulkan.lunarg.com/doc/sdk/1.1.121.1/mac/getting_started.html

But, is this where the RealSense SDK expects it to go once installed?

Regards,

Steve.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

Thank you for reporting the issue. In regard to the OSX installation issue, it should be addressed by PR #5264.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Please also refer to the following thread for errors about factory.h.
https://github.com/IntelRealSense/librealsense/issues/4186
Thank you.

closed 2019-12-11 by RealSenseCustomerSupport