realsenseai/librealsense #4978: rs-kinfu Build Issue - Unresolved Exernal Symbol glfw3.lib

Veng3nce, 2019-10-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.29.0                          | 
| Language                            |  C/C#/opencv    | 

### Issue Description
I am trying to build the kinfu opencv example but can't seem to get it to work. I have tried to fix this issue but haven't found a solution so far.

[image]

Does anyone know a fix for this issue? Thanks.

Build Steps:

1) Built opencv 4.1.1 and opencv contrib using cmake
2) Built and installed GLFW 3.3 using cmake
3) Built librealsense using cmake
[image]

Note: Built using VS 2019 for x64.

I tried the following:
[add library and include path](https://stackoverflow.com/questions/20223198/visual-studio-11-glfw-external-symbol-error)
[add libraries to input](http://discourse.glfw.org/t/link-error-with-glfw/382)

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

@Veng3nce You issue seems very similar to https://github.com/IntelRealSense/librealsense/issues/4278 Could you please try the solution there? Thanks!

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

@Veng3nce Could you please update? Thanks!

Veng3nce, 2019-11-20:
@RealSenseCustomerSupport I've currently moved on from this project, will be happy to update the ticket once I have tried the solution in #4278.

closed 2019-11-20 by Veng3nce