realsenseai/librealsense #4979: No Motion data from D435i on Raspberry Pi 4

AnupamYedida, 2019-10-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i   | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Debian Buster   | 
| Kernel Version (Linux Only)    |  4.19.75              | 
| Platform                                 | Raspberry Pi4  |
| SDK Version                            |   2.28.0                           | 

The depth and RGB data is fine. But there is no IMU(Gyro and Accel) data being published. It returns Error type: No data to return (-21).
Bus error.
I have a desktop running Ubuntu 18.04 and the motion data is published over there perfectly.

MartyG-RealSense, 2019-10-04:
Another Pi 4 user has reported this error with the D435i recently, though the customer stopped responding before the case was resolved.  I provide the link below as a reference regarding what possible causes were looked at.

https://forums.intel.com/s/question/0D50P00004RoUHqSAN/d435i-motion-module-force-pause-when-turning-on-imu?language=en_US

AnupamYedida, 2019-10-04:
I have already gone through this link and followed along. Sadly no progress with that :(

MartyG-RealSense, 2019-10-04:
It could be part of other recent issues relating to Pi4 / Raspbian Buster.  Examples:

https://github.com/IntelRealSense/librealsense/issues/4899

https://forums.intel.com/s/question/0D50P00004RoPrGSAV/rpi-4-raspbian-buster-realsenseviewer-cannot-detect-d435i?language=en_US

https://github.com/IntelRealSense/librealsense/issues/4565#issuecomment-518359584

AnupamYedida, 2019-10-04:
Tried that link as well, unfortunately no luck with that. I would appreciate any help with this.

AnupamYedida, 2019-10-04:
Can it be a possible kernel issue? I am running 4.19.75

MartyG-RealSense, 2019-10-05:
Regarding the question of whether kernel 4.19 is supported by Librealsense, whilst it is not officially, a Pi 4 user reported it worked for them with Raspian Buster.

https://github.com/IntelRealSense/librealsense/issues/4818#issuecomment-529195542

AnupamYedida, 2019-10-06:
I suspect it 4.19 has issues with the HID interface. There are kernel patches for Ubuntu 16 and 18, but I couldn't get it to work on Debian Buster on the Pi4.

ev-mp, 2019-10-10:
@AndreyIvanovValgo hello, we've recently  received RPI4 units to start working on the reproduction.
One immediate observation is that running IMU sensors using the native Raspbian kernel is not possible as the core IIO kernel modules are not built into it. So for full support one will need to reconfigure and recompile the kernel, which is not, obviously, the first choice.

We'll look into the reasons for `FORCE_LIBUVC=true` not working properly as it should and update accordingly.
Sorry for the inconvenience

AnupamYedida, 2019-10-10:
I see. Much appreciated :)

rofreli, 2019-10-12:
Thank you, MartyG-Realsense, that's mean I will continue forcing with ubuntu in raspberry pi 4. It does not work well, but I can test somethings, including IMU.

RealSenseCustomerSupport, 2021-08-14:
----------------------------------------------

@AnupamYedida is this issue still occurring with the latest SDK and f/w combinations?

RealSenseCustomerSupport, 2021-08-20:
----------------------------------------------

@AnupamYedida it seems that this issue has been resolved so if we do not hear otherwise we will close this ticket next week.

closed 2021-09-09 by RealSenseSupport