realsenseai/librealsense #4980: Is disabling relocalization possible in the new releases?

karry3775, 2019-10-04:
I am using a intel t265 tracking camera. I wanted to disable the relocalization present in the device. https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md says that RS2_OPTION_ENABLE_RELOCALIZATION is present to do that. However I cannot locate it in any of the launch files.

MartyG-RealSense, 2019-10-04:
The link below has information on options for pose jumping and relocalization that can be set to disabled.

https://github.com/IntelRealSense/librealsense/commit/de76d3a1bdd094100a3985e4dd3273e8e1fd2bb6

In regard to ROS, a user suggested what they thought would be a better approach than relocalization.

https://github.com/IntelRealSense/realsense-ros/issues/779#issuecomment-494046537

karry3775, 2019-10-08:
> de76d3a
> IntelRealSense/realsense-ros#779 (comment)

Thanks I will try this out and get back to you.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi karry,

Were you finally able to disable relocalization?

closed 2019-11-14 by RealSenseCustomerSupport