realsenseai/librealsense #4982: Saving image as png using imwrite

hojunson, 2019-10-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {opencv }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am trying to save image streams and IMU using callback function. 
I am using threads to save data without any loss. 
Interestingly, some of color images I got look like
[image]

Some of them is good but I got weird images(above case there is unsynched area at the line) such like that when I saved data while moving.
Is there any suggestion?

kjacks21, 2019-10-07:
I haven't worked with the IMU and not sure what language you are working in, but #4934 has code to save the color frames as .png, depth as matrices, and to not have frame drops.

g2-bernotas, 2019-10-07:
You can save depth frames as 16-bit png. Here is C++ code:

``` C++
std::vector<int> compression_params;
compression_params.push_back(CV_IMWRITE_PNG_COMPRESSION);
compression_params.push_back(0); // lossless, the number is 0-9, the most data will be lost at 9 (lightest), none at 0 (heaviest)
 
std::string dateTime;
timeStamp(dateTime, "");
 
cv::imwrite(dateTime + ".png", image_full, compression_params); // save filename as date
cv::Mat image_full_saved = cv::imread("fileName.png", CV_LOAD_IMAGE_ANYDEPTH); 
```

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @hojunson Do you have further question on this?

hojunson, 2019-10-30:
No. Closing is fine Thanks.

closed 2019-10-30 by RealSenseCustomerSupport