realsenseai/librealsense #4983: 3D Area(Map) Reconstruction using both D435(i) and T265

sadransh, 2019-10-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 and D435 } | 
| Firmware Version                   | (  0.1.0.279 and 05.11.15) | 
| Operating System & Version |   {Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  (4.15.0.65-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.29.0 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Is it possible to reconstruct an area in real-time with combination of these two cameras? 
If yes, should I use Open3D to join pointclouds ( or it is just neccessary for high-accuracy?)? 
Is it possible to use Open3d in real time ( with considering a highend pc or high end embedded board(like jetson TX2) ? 

In other words, I want to do sth like [this](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i), However, instead of using RTabMap and feature match , using tracking data from T265 ( to have maintain higher moving speed and accuracy in reconstructed 3d map  and less processing on main processor). 

Can anyone tell how much does adding the T265 for map reconstruction, help accuracy and fater robot movements? (in comparison with the above mentioned link)

MartyG-RealSense, 2019-10-05:
Map construction with SLAM can be done with D435i on its own, as described in the Intel guide in the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i?language=en_US

In regard to pairing T265 and D435i together: you can do a pairing with the standard D435, as the T265 has the same IMU component as the D435i (though they have some differences in how they process the IMU data).  The official documentation says though that D435i can be paired with T265 if the D435i is your preferred choice of depth camera.

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#should-i-use-a-t265-in-addition-to-or-instead-of-the-d435i

The official FAQ for the T265 actually suggests that the D415 model may be the optimum pairing for best depth quality (presumably as the D415 has around half the RMS error (error over distance) of the D435 models).   A D435 model may be best for you though if your robot is going to be fast-moving, as those models have a faster 'global shutter', whilst the D415 has a slower 'rolling shutter'.

An excellent reference for pairing T265 and D435 is a long YouTube seminar presentation about T265, which illustrates the pairing at 17 minutes 15 seconds in.

https://youtu.be/62vm0_RZ1nU?t=1036

I cannot find any information about using T265 with Open3D, so cannot make any comment unfortunately on that subject.   One of the Intel staff on this forum may be able to add some insights on that matter.

In regard to how T265 aids sensing during fast movements:  I am not a T265 specialist and so can only comment about generalities - there are people on the RealSense team with far more specialist knowledge of it.  The link below may be useful reading though.

https://github.com/IntelRealSense/librealsense/issues/3923

tsbertalan, 2019-10-05:
The approach is taken in the supplied launch files in the realsense-ROS
sibling repository already uses the T265 for odometry instead of the
software-based visual odometry that is the default in RTABMAP. This has
worked for me with an i5 CPU and a D435 (no IMU), with fairly minimal
modification of the supplied launch file.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi @sadransh - Were you able to review the information provided by MartyG-RealSense and tsbertalan?  Is there anything else needed on this topic?

sadransh, 2019-11-05:
Yes, I, as @tsbertalan, said there was the sample provided in rs-ROS works well. However, for some reason, I have stopped that thread of my work. I think it's better to close the issue for now. 

Thanks to @MartyG-RealSense for provided information, too.

closed 2019-11-05 by sadransh