realsenseai/librealsense #4984: Are there any multiple cameras code examples with python language?

Dontla, 2019-10-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435&D435i| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python                           | 
| Segment			|  Robot                       | 

### Issue Description
I have read the python wrapper examples here ([https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python#python-wrapper](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python#python-wrapper)) and the python wrapper here ([https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html#module-pyrealsense2](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html#module-pyrealsense2)), but I still don't know how to deal with my cameras...

MartyG-RealSense, 2019-10-05:
There is a Python example called **'box_dimensioner_multicam' that supports multiple cameras for the purpose of measuring the volume of objects placed upon a chessboard image.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

Dontla, 2019-10-16:
Grazie!

closed 2019-10-16 by Dontla