realsenseai/librealsense #4985: D400 modules with custom RGB sensor

Jiloc, 2019-10-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 

Hello, my research group needs to reach a higher accuracy with the RGB images. We noticed that all modules, except the D415, comes without a RGB module, so it should be possible to connect an external camera to the others. In particular we want to use [this camera](http://www.mindvision.com.cn/cpzx/info_62.aspx?itemid=1445&lcid=101) (or something similar). If it's possible, has it ever been done before? Is there a mechanism to synchronize the RGB and the Depth modules? More in general, are there any preconditions or specifications to do it?

MartyG-RealSense, 2019-10-05:
Yes, only the D415 Depth Module board has an integrated RGB sensor.  In the D435 cameras (which contain an RGB sensor), the RGB sensor is connected separately via a cable.   The RGB sensor can be mounted on the end of the D430 depth module.

[image]

The image below gives a rear view of the RGB module and its cable.  I believe it attaches to the second 50 pin connector on the Vision Processor D4 (the other 50 pin connector being occupied by the depth module cable).

[image]

The above image came from a Chinese website that did a teardown analysis of the D435i.

https://ek21.com/news/tech/17657/

The link below has a recent case of a RealSense user doing a more complex build involving building their own Vision Processor D4 setup from its individual components so they could integrate it into their unique system board.

https://support.intelrealsense.com/hc/en-us/community/posts/360036002353

Prior to March 2018, Intel planned to allow users to connect custom RGB sensors to the second D4 connector.  Those plans were dropped.  So the physical pin connections for adding your own sensor via a custom cable are there, but I'm not sure the firmware support for a non-Omnivision RGB sensor is.  

It was originally thought that if you added your own RGB sensor then all that would be required afterwards would be to do a re-calibration of the camera to get the new sensor working.  After the custom sensor plans were dropped though, I do not know if that is still the case.

The pin-outs for the second connector on the Vision Processor D4 are available in the January 2018 edition of the 400 Series data sheet, in the Tracking Module section on pages 50-55.  The tracking module section was removed from the data sheet after March 2018 when the T150 Tracking Module component intended for the connector was retired by Intel.

https://www.rutronik.com/fileadmin/Micropages/Intel/Intel_RealSense_D400_Series_Datasheet.pdf

Jiloc, 2019-10-05:
Hello Martin and thanks for the prompt reply. 

I should conclude that it "should" be possible to use a custom camera (with our requirements), but any information about it is undisclosed. Am I right?

MartyG-RealSense, 2019-10-05:
I'm not sure it's a secret, but rather that not many people have tried it and so there is not a lot of information on it in the public domain to refer to.  Intel themselves certainly create a range of RealSense camera configurations for certain customers that are not available for public retail.

If you need further specific technical information that is not in the public domain then I can put you in touch with an Intel Support manager to start you on the path to registering a non-disclosure agreement (NDA).

Jiloc, 2019-10-05:
That would be great!

MartyG-RealSense, 2019-10-05:
Ok, if you leave your email address in a comment below then I will send you the Intel manager's direct email (I have his permission for such referrals).  Once I have the address, I will let you know and you can edit out your email address if you wish so it stays private.

Jiloc, 2019-10-05:
francescoleacche@gmail.com

MartyG-RealSense, 2019-10-05:
Ok, I've got it.   I'm mailing you the details now.

rpardell-edgebrain, 2019-10-07:
Hi, we have the same issue here. We have been using the D435 but the OV2740 sensor has a very poor sensitivity. We would like to use a MIPI RGB module based on the OV2770 sensor, having the same resolution and general characteristics of the OV2740 but offering 2.8 micron pixels with much better sensitivity. Could we join forces here?

MartyG-RealSense, 2019-10-07:
If you'd like, I can direct you to a manager who can start you on the path to NDA registration.  If you would like to do this, could you tell me what country are you in please so I can direct you to the appropriate regional manager?

rpardell-edgebrain, 2019-10-07:
Hi, that would be fantastic! We are UK based.
Please send NDA and regional manager details to ricard@earthrover.farm

MartyG-RealSense, 2019-10-07:
I will forward your details now to Milo, the EMEA (Europe, Middle East and Africa) technical manager whose territory covers the UK.  Then he will contact you directly at the provided address.   I have the address now, so feel free to edit it out if you want to keep it private.

MartyG-RealSense, 2019-10-09:
Miro spoke to me about your case and similar ones today.  He said that unless a customer is likely to bring a large 'return on investment' to Intel in excess of a huge volume of "100,000 a quarter" then Intel could not provide specialist NDA-level support, unfortunately.   The research group of @Jiloc above may be in the same situation.

closed 2019-11-01 by RealSenseCustomerSupport

rpardell-edgebrain, 2019-11-01:
We never received the required contact. Could you reopen this issue please?

Enviat des del meu iPhone

Ricard Pardell
 
EDGEBRAIN
Artificial Intelligence at the Edge
mailto:rpardell@edgebrain.io
Telephone: +34 626 243 989
Skype: ricard.pardell

Carrer Segre, 2
08174 Sant Cugat del Vallès
Barcelona
Spain

http://www.edgebrain.io
 
Unless otherwise stated, this email and its attachments are confidential and proprietary.

MartyG-RealSense, 2019-11-01:
@rpardell-edgebrain   It is unlikely that Intel will be able to assist with attaching a custom RGB sensor unless a large-volume order for a camera or some other significant benefit to them was involved.

A more acievable route would be to explore syncing RealSense depth with the RGB of an external camera, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/4202