realsenseai/librealsense #4986: Please add -latomic to the ARM build

raymondbedwards55, 2019-10-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.6100 | 
| Operating System & Version |   /etc/debian_version=10.1  | 
| Kernel Version (Linux Only)    |   4.19.66-v7+                                         | 
| Platform                                 | Raspberry Pi 3b+  |
| SDK Version                            | librealsense commit 81d469                    | 
| Language                               |  python                         | 
| Segment			        |  ?what?                          | 

### Issue Description

After downloading the source for the python wrapper and compiling:

    $ python -c "import pyrealsense2"
    ImportError: /usr/local/lib/librealsense2.so.2.28: undefined symbol: __atomic_fetch_add_8

The fix was to add a `-latomic` to the gcc command creating librealsense2.so.2.28 so ldconfig knew it depended on libatomic.  Could you consider adding the fix please?

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi @raymondbedwards55 - Have a look at the pull request link provided here.  I think this should take care of your request.

https://github.com/IntelRealSense/librealsense/pull/5025

dorodnic, 2019-11-11:
Should be resolved now

closed 2019-11-19 by dorodnic