realsenseai/librealsense #4988: Depth Image's Size Reduced After Postprocess

riven314, 2019-10-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | Lenovo Legion Y740 |
| SDK Version                            |  v2.25  | 
| Language                            |  python    | 
| Segment			|  depth filtering    | 

### Issue Description
I tried to apply depth filtering on realsense image of resolution (1280, 720). But I found that after filtering, the depth image I received have a lower resolution (i.e. (640, 360)). How could I preserve the resolution of depth image after filter post-processing?

### My Code Snippet for Depth Post-processing

    def filter_depth(depth_frame):
        # setup filter processes
        decimation = rs.decimation_filter()
        depth_to_disparity = rs.disparity_transform(True)
        spatial = rs.spatial_filter()
        temporal = rs.temporal_filter()
        disparity_to_depth = rs.disparity_transform(False)
        hole_filling = rs.hole_filling_filter()
        # apply the filters
        frame = decimation.process(depth_frame)
        frame = depth_to_disparity.process(frame)
        frame = spatial.process(frame)
        frame = temporal.process(frame)
        frame = disparity_to_depth.process(frame)
        frame = hole_filling.process(frame)
        return frame

MartyG-RealSense, 2019-10-06:
It is likely the decimation filter that is halving your resolution.  The link below is to a case where the same thing was happening to someone, and an explanation is provided.

https://github.com/IntelRealSense/librealsense/issues/1284

riven314, 2019-10-06:
Thanks for the prompt reply!
I will close the issue now, understanding that the filtering process "decimation" will cause (exponential) size reduction for depth image

closed 2019-10-06 by riven314