realsenseai/librealsense #4990: realsense-viewer warnings, depth no longer working, and out of date udev rules

kjacks21, 2019-10-06:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Linux 5.0.0-31-generic|
|**Name**|Intel RealSense D435|
|**Serial Number**|834412071708|
|**Firmware Version**|05.11.15.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|836313021935|
|**Firmware Update Id**|836313021935|

**Issue Description**
I've commented on other issues that seemed relevant, but now I'm not so sure. Hence, I'm bringing everything here. Realsense-viewer and depth capture previously worked fine, but as of a month or so ago (I don't recall exactly because I noticed the problem only a few days ago) it no longer does. My main issue is that I am unable to extract depth data from the .bag files that I've generated recently, and I suspect it has to do with the following warnings and errors.

**Related**
Potentially related #4971, #4951; however I have a newer kernel version. #4957 has the closest specs as me and as I mentioned there, I tested the camera with the new firmware version on a different computer that was previously working with an old SDK and it doesn't work there.

**Trace**
Upon starting up `realsense-viewer` without a camera plugged in, I get the following warnings:

```
 06/10 14:19:19,642 ERROR [140075786561280] (types.h:304) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 06/10 14:19:19,643 WARNING [140075786561280] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 06/10 14:19:19,650 ERROR [140075786561280] (types.h:304) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 06/10 14:19:19,650 WARNING [140075786561280] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
```
Plugging the camera in, I get the following:

```
06/10 14:21:53,269 WARNING [140076051208064] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
```

In addition, I am given the following warning:

```
Timestamp: 1570385959626.153564
Severity: Warn
Description: RealSense UDEV-Rules file:
 /lib/udev/rules.d/60-librealsense2-udev-rules.rules
 is not up-to date!
UDEV-Rules permissions configuration 
 for RealSense devices.

Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors
unless the application is running under 'sudo' (not recommended)
In case of Debians use: 
sudo apt-get upgrade/install librealsense2-udev-rules
To manually install UDEV-Rules in terminal run:
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger
```
which does not resolve itself when upgrading and installing librealsense2-udev-rules. See console output below:

```
librealsense2-udev-rules is already the newest version (2.29.0-0~realsense0.1579).
```

In the viewer, when I turn on the Stereo module, I can see the stream but get the following:

```
 06/10 14:27:19,288 WARNING [139927887017728] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
```

Similarly, when I turn on the RGB Camera, I get the following:

```
06/10 14:28:26,942 WARNING [139927870232320] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
```
[... 68 lines cut ...]
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Tm2
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Software Device
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Software Sensor
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Decimation Filter
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Threshold Filter
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Disparity Filter
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Spatial Filter
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Temporal Filter
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Hole Filling Filter
 06/10 14:29:10,333 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Zero Order Filter
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Pose
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Pose Sensor
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Wheel Odometer
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Global Timer
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Updatable
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Update Device
 06/10 14:29:10,334 WARNING [139927855548160] (record_device.cpp:255) Extensions type is unhandled: Tm2 Sensor
```
I am able to view both streams from the recorded .bag file in the viewer; however, when I try to access the depth frames from the .bag file using the API, I get an error. For example, the trace below:

```
 depth_sensor = profile.get_device().first_depth_sensor()
RuntimeError: Could not find requested sensor type!
```
Please let me know if this needs to be broken up into multiple issues. Thanks fort he support!

kjacks21, 2019-10-27:
Updated title to show that updating pyrealsense2 resolved a portion of the issue.

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @kjacks21,

1. Please see #4586 with the similar issue since the kernel v5.0 is not yet supported. That's the reason you saw some warnings. You may need to manually apply the patches.
2. what's your "pyrealsense2" version?
3. You should be able to playback your bag file thru realsense-viewer, right?

kjacks21, 2019-10-29:
@RealSenseCustomerSupport thanks for the response! `pyrealsense2` version is 2.29.0.1124, but was 2.22 when it wasn't working. Correct, I am able to play back. Since this can be resolved with #4586, I'll close this issue.

closed 2019-10-29 by kjacks21