realsenseai/librealsense #4992: Q: Is there a reason for the specific USB cable provided with the T265?

neilyoung, 2019-10-06:
... I'm asking, because I found that USB 2 seems to be sufficient at the first glance.

MartyG-RealSense, 2019-10-06:
Cables supplied with RealSense cables are carefully validated by Intel to work well with the camera.  

The official T265 documentation page says that whilst pose data can be streamed on USB2, fisheye data requires USB3.  The cable supplied with the T265 would therefore have to be capable of providing the device's full range of modes.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#what-are-the-system-requirements

neilyoung, 2019-10-07:
Thanks. Very quick, super clear :)

RealSenseCustomerSupport, 2019-10-11:
----------------------------------------------

Thanks @MartyG-RealSense for the information.

closed 2019-10-11 by RealSenseCustomerSupport