realsenseai/librealsense #4993: How to use Advanced Control  on C++/CLI

kazu0622, 2019-10-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 0.5.11.01.100 | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |                                     | 
| Platform                                 | PC  |
| SDK Version                            |  {  2.20.0 }                          | 
| Language                            |  {C/C#}                          | 
| Segment			|  {others }                          | 

I'm tring to use Advenced_mode control toward D435. The C++ description I wrote is as below:
rs2::pipeline_profile selection = pipe->start(*cfg);
rs2::device selected_device = selection.get_device();
//Advanced Mode
auto advanced_mode_dev = selected_device.as<rs400::advanced_mode>();
const int current_value_mode = 0;//current value
auto depth_control_group = advanced_mode_dev.get_depth_control(current_value_mode);
depth_control_group.deepSeaSecondPeakThreshold = (uint32_t)1023;
advanced_mode_dev.set_depth_control(depth_control_group);

It works well on pure C++ code with VisualStudio2015.
But , Building process failed when I wrote this under C# wrapper description on C++/CLI as below.

C:\Program Files (x86)\Intel RealSense SDK 2.0\include\librealsense2/rs_advanced_mode.hpp(21): error C2679  no operator'=' found which takes a right-hand operand of type ''nullptr';

Could you give me some advice to cope with this issue?

Thank you.

MartyG-RealSense, 2019-10-07:
As far as I am aware, the C# wrapper does not have access to Advanced Mode commands.  An alternative might be to define a 'json' configuration file of camera settings and load that into your project with C# code.

https://github.com/IntelRealSense/librealsense/issues/3275#issuecomment-464723090

kazu0622, 2019-10-07:
@MartyG-RealSense 
Thank you for your quick response.
Is my understanding correct that Advanced Mode commands can't be accessed  through C++/CLI even if I describe C++ not C#?

MartyG-RealSense, 2019-10-07:
I do not have knowledge of C++/CLI unfortunately and there seems to be very little information available on it other than this link:

https://iamzombie.tistory.com/51

If you were trying to access Advanced Mode through the C# wrapper then it would not be able to do so if those commands were not exposed to the end-user in the C# wrapper.

If you were accessing C++ instead of the C# wrapper, I would have thought that Advanced Mode commands should work.

An alternative in the C# wrapper to using C++/CLI is apparently something called **Platform Invoke**.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/pinvoke.md

I apologize that I do not have the knowledge of the subject to provide a more definitiive answer.   One of the Intel RealSense staff on this forum are likely to be able to advise better on this particular question.

kazu0622, 2019-10-07:
@MartyG-RealSense 
Thank your for your advice. Let me change my question.
Can I set the Preset mode from Default to  High Accuracy easier than controling Advanced mode?
If so ,could you give me  some example C++ code?
The reason why  I wanted to control Advanced mode(DS Second Peak Threshold) is getting accurate depth data.

Thank you.

MartyG-RealSense, 2019-10-07:
Apologies for the delay in responding further, I was carefully researching your new question.

There are Low, Medium and High resolution versions of the High Accuracy preset.  So I would try substituting '1.json' in the earlier script for one of these names:

**High**
HighResHighAccuracyPreset.json

**Medium**
MedResHighAccuracyPreset.json

**Low**
LowResHighAccuracyPreset.json

It looks s though the parameter 'secondpeakdelta' sets DS Second Peak Threshold, though I could not find C++ code to do so.  Hopefully someone on this forum can provide code for that.

https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/advanced_mode/json_loader.hpp#L297

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

@kazu0622 If you want to load json file with the parameter 'secondpeakdelta' DS Second Peak Threshold, I think you can use RealSenseViewer to adjust this parameter and save the configuration to your own json file.  Below is one json file example after the modification of this parameter.  Also I attached one code snippet to enable advanced control and load json file in https://gist.github.com/sherrysun2018/61c94dcbe63f02ce7bad159748d196f2
Please have a try to see if you have any other questions.

{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "1",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "751",
    "param-lambdacensus": "6",
    "param-leftrightthreshold": "10",
    "param-maxscorethreshb": "2893",
    "param-medianthreshold": "796",
    "param-minscorethresha": "4",
    "param-neighborthresh": "108",
    "param-raumine": "6",
    "param-rauminn": "3",
    "param-rauminnssum": "7",
    "param-raumins": "2",
    "param-rauminw": "2",
    "param-rauminwesum": "12",
    "param-regioncolorthresholdb": "0.785714",
    "param-regioncolorthresholdg": "0.565558",
    "param-regioncolorthresholdr": "0.985323",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "0",
    "param-robbinsmonrodecrement": "25",
    "param-robbinsmonroincrement": "2",
    "param-rsmdiffthreshold": "1.65625",
    "param-rsmrauslodiffthreshold": "0.71875",
    "param-rsmremovethreshold": "0.809524",
    "param-scanlineedgetaub": "13",
    "param-scanlineedgetaug": "15",
    "param-scanlineedgetaur": "30",
    "param-scanlinep1": "155",
    "param-scanlinep1onediscon": "160",
    "param-scanlinep1twodiscon": "59",
    "param-scanlinep2": "190",
    "param-scanlinep2onediscon": "507",
    "param-scanlinep2twodiscon": "493",
    "param-secondpeakdelta": "370",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "1722",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

@kazu0622 Any other questions about this ticket? Thanks!

kazu0622, 2019-11-19:
Thank you for all of  your comment.

closed 2019-11-19 by kazu0622