realsenseai/librealsense #4994: Imu frame drop and freeze from enable_device_from file

hojunson, 2019-10-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) /  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am trying to save all data from D435i(depth, color, and IMU)
through the realsense viewer, rosbag file was generated. 
then using enable_device_from_file to save all data separately, there was frame drop and freezing. and I am using set_real_time(false) 
In case of poll_for_frame(), imu frame was dropped, and with callback
freezing has occurred. 
When I check the .bag file with the Viewer, it looked fine. Fps of imu was 62.8 for accelerator and about 200 for gyro, which I set up to record.

Is there any suggestion?

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @hojunson,

You did try and modify below rs-callback example to playback your rosbag file and see the IMU frame drop and freezing issue?
https://github.com/IntelRealSense/librealsense/tree/master/examples/callback
which librealsense and firmware version do you use?

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@hojunson,

Don't see any response for a while. If no further question, will close this.

hojunson, 2019-12-06:
Yes please

closed 2019-12-06 by RealSenseCustomerSupport