realsenseai/librealsense #4995: cannot glfw-imgui.lib

devLupin, 2019-10-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Windows 10  |                                    | 
| Platform                                 | PC |
| SDK Version                            |  2.0 |                          | 
| Language                            |  {C++ }                          | 
| Segment			|  Gait analysis}                          | 

### Issue Description

I ran "multicam" in the sample code. There was a link error. The contents are as follows.

"LINK : fatal error LNK1104: 'C:\Program Files (x86)\Intel RealSense SDK 2.0\samples\multicam\Debug\glfw-imgui.lib' cannot open file."

The file doesn't really exist. 

My actions to fix the error are:
<project properties>
1. vc++ directory -> include directory ==> C:\Program Files (x86)\Intel RealSense SDK 2.0\third-party\glfw-imgui\include
2. linker->general ==>C:\Program Files (x86)\Intel RealSense SDK 2.0\lib\x86;C:\Program Files (x86)\Intel RealSense SDK 2.0\lib\x64
3. linker -> input -> Additional dependencies ==> realsense2.lib

No source code modifications.

That's all. somebody help me!!

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

Hi @devLupin,

Please make sure that you follow Windows 10 Installation instructions that described here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md
After completion of this process, please run "multicam" example and let us know if you still
experience the same issue.

Thank you!

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

@devLupin,

Do you still need help with this question?

devLupin, 2019-11-02:
@RealSenseCustomerSupport 

I solved it in a different way. thank you!

closed 2019-11-06 by RealSenseCustomerSupport