realsenseai/librealsense #4996: realsense-viewer T265 Ubuntu 16.04: Warning at startup

neilyoung, 2019-10-07:
I'm getting this warning while starting realsense-viewer in Ubuntu 16.04.

```
Timestamp: 1570439512791.595703
Severity: Warn
Description: RealSense UDEV-Rules file:
 /lib/udev/rules.d/60-librealsense2-udev-rules.rules
 is not up-to date!
UDEV-Rules permissions configuration 
 for RealSense devices.

Missing/outdated UDEV-Rules will cause 'Permissions Denied' errors
unless the application is running under 'sudo' (not recommended)
In case of Debians use: 
sudo apt-get upgrade/install librealsense2-udev-rules
To manually install UDEV-Rules in terminal run:
$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger

```

Checking permissions on that file gives:

```
decades@ubuntu:~$ ls -lall /lib/udev/rules.d/60-librealsense2-udev-rules.rules
-rw-r--r-- 1 root root 6451 Sep 23 07:50 /lib/udev/rules.d/60-librealsense2-udev-rules.rules

```

Trying to run the suggested command gives:

```
decades@ubuntu:~$ sudo cp ~/.99-realsense-libusb.rules /etc/udev/rules.d/99-realsense-libusb.rules && sudo udevadm control --reload-rules && udevadm trigger
cp: cannot stat '/home/decades/.99-realsense-libusb.rules': No such file or directory

```

Any suggestion how to get rid of this warning?

TIA

neilyoung, 2019-10-07:
Possible duplicate of https://github.com/IntelRealSense/librealsense/issues/4990

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

Hi neilyoung,

I have confirmed on Ubuntu with latest librealsense2  Ver. 2.30.0 release that this warning no longer happens.

Will you be needing any further assistance on this?

Thanks

neilyoung, 2019-11-19:
Thanks. Can be closed.

closed 2019-11-19 by RealSenseCustomerSupport