realsenseai/librealsense #4997: realsense SR300 doesn't work without internet

Lee-daeho, 2019-10-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   |pyrelasense 2.2.1| 
| Operating System & Version |  UBUNTU 16.04  | 
| Kernel Version (Linux Only)    | 4.4.xx                                          | 
| Platform                                 | NVIDIA JETSON TX2 |
| SDK Version                            |                        | 
| Language                            |  PYTHON                          | 
| Segment			|  CUDA ERRORS                        | 

### Issue Description
Hi i'm using SR300 model on Jetson TX2.
as i write on title, it doesn't work without internet.
when i'm using it on offline, it makes error that on function ' void rscuda::unpack_yuy2_cuda_helper(const uint8_t* h_src, uint8_t* h_dst, int n, rs2_format format) ',
assert(result==cudasuccess):failed.
but it becomes fine when i connect internet. is this normal thing? or can i fix it? i should use realsense on offline situation, so I need some help to fix it. 

Thanks.

p.s. i swear it's fine when i do my job on internet. i have no idea why it doesn't work.

ev-mp, 2019-10-22:
@Lee-daeho , can you share the generated logs ?
Running the SDK does not require any internet connection. The steps that need it are `git clone` and `cmake`. We haven't received any reports of this kind so far so it is a rather odd observation. 

The only thing that I could think of is that when installing the NVidia's SDK manager there was some kind of "user participation/experience" checkbox marked, and now when you try to activate NVidia's cores then the drivers tries to report statistics back to NVidia's server. But this is a pure conjecture.

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Hi @ Lee-daeho,

Any update? Is the issue clarified?

Lee-daeho, 2019-10-30:
Sorry for late. Nowadays it doesnt matter if it is connected to internet or not.
here is my error log.

2019-10-30 19:30:31.055716: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:864] ARM64 does not support NUMA - returning NUMA node zero
2019-10-30 19:30:31.055893: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1392] Found device 0 with properties: 
name: NVIDIA Tegra X2 major: 6 minor: 2 memoryClockRate(GHz): 1.3005
pciBusID: 0000:00:00.0
totalMemory: 7.66GiB freeMemory: 4.71GiB
2019-10-30 19:30:31.055968: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1471] Adding visible gpu devices: 0
2019-10-30 19:30:33.546964: I tensorflow/core/common_runtime/gpu/gpu_device.cc:952] Device interconnect StreamExecutor with strength 1 edge matrix:
2019-10-30 19:30:33.547072: I tensorflow/core/common_runtime/gpu/gpu_device.cc:958]      0 
2019-10-30 19:30:33.547117: I tensorflow/core/common_runtime/gpu/gpu_device.cc:971] 0:   N 
2019-10-30 19:30:33.547358: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1084] Created TensorFlow device (/job:localhost/replica:0/task:0/device:GPU:0 with 4096 MB memory) -> physical GPU (device: 0, name: NVIDIA Tegra X2, pci bus id: 0000:00:00.0, compute capability: 6.2)
python3: /home/nvidia/librealsense/src/cuda/cuda-conversion.cu:273: void rscuda::unpack_yuy2_cuda_helper(const uint8_t*, uint8_t*, int, rs2_format): Assertion `result == cudaSuccess' failed.
Aborted (core dumped)

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Hi @Lee-daeho,

It's great that the issue is resolved. As @ev-mp mentioned, running librealsense apps with SR300 should work without internet. But, during the SDK configuration and building stage, the internet is required due to downloading some packages/firmwares from internet.

If no further question for this topic, will close this.