realsenseai/librealsense #4998: Some Problem When Run odroid patch script

xilunaJumma, 2019-10-07:
Hi,Please help me to resolve these problem.
When I run "./scripts/patch-realsense-ubuntu-odroid.sh"
./scripts/patch-realsense-ubuntu-odroid.sh: line 97: ../scripts/realsense-camera-formats_ubuntu-xenial-hwe.patch: No such file or directory

When I run another file "./scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh"
./scripts/patch-realsense-ubuntu-odroid-xu
[image]
4-4.14.sh: line 106: ../scripts/realsense-camera-formats_ubuntu-bionic-master.pat        ch: No such file or directory

I really not find these two file in folder /scripts.
[image]

ev-mp, 2019-10-10:
@xilunaJumma hello, the patches names have been recently updated, so the requested patch's name is, I believe, `realsense-camera-formats-bionic-master.patch`.
We provide hosting but do not maintain community-provided scripts, so unfortunately those may gradually become outdated.

In case you manage to update and build successfully I would ask you to consider to submit a fixing PR so that this info will be shared with the community.

xilunaJumma, 2019-10-12:
Yes, i have verify the name is realsense-camera-formats-bionic-master.patch.  
But there is not only one mistake in in this file,I will update my script when I really test successfully.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@xilunaJumma Thanks for the input and did you get it through? Looking forward to your update. Thanks!

xilunaJumma, 2019-10-31:
I really got it through,but I will update about a week later. 🤗-------- 原始邮件 --------发件人： RealSense Customer Support <notifications@github.com>日期： 2019年10月31日周四 中午12:27收件人： IntelRealSense/librealsense <librealsense@noreply.github.com>抄送： xilunaJumma <lwma@xiluna.com>, Mention <mention@noreply.github.com>主    题： Re: [IntelRealSense/librealsense] Some Problem When Run odroid patch script (#4998)
@xilunaJumma Thanks for the input and did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@xilunaJumma What's the status now? Saw you were saying more mistakes in previous reply. Did you get it through now? Thanks!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@xilunaJumma Could you please update. If we don’t hear from you in 7 days, this issue will be closed. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@xilunaJumma  Sorry we didn't hear from you for weeks. Will close the ticket at this point. Please feel free to create another ticket if you still have issue or questions. Thanks!

closed 2020-02-17 by RealSenseCustomerSupport