realsenseai/librealsense #5000: Windows 7 drivers missing from installers after 2.19.0 and newer

girrr, 2019-10-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.10.00 | 
| Operating System & Version |  Windows 7 | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.0 and newer                          | 
| Language                            |  N/A                          | 
| Segment			|  others                         | 

### Issue Description
It seems that the drivers folder is missing from any installer after 2.18.1.

Is this intentional or is it actually missing?

dorodnic, 2019-10-24:
Will be fixed in upcoming release

dorodnic, 2019-11-11:
Should be resolved now

closed 2019-11-19 by dorodnic