realsenseai/librealsense #5004: T265 precision over time

BriceRenaudeau, 2019-10-08:
Hello, 
I am using the T265 on an indoor mobile robot using ROS. I am using EKF to fuse both visual odometry from the T265 and wheel odometry.
[Ubuntu 18.04.3, ROS Melodic, Last camera firmware]
Some month ago I made several records of the values sent by the camera and the result where in harmony with the values sent by the wheels:
[image]
[image]

But now the values are overestimated or underestimated by up to 20 %:
[image]
[image]
I tried to use another T265 camera, but the result is similar.
[image]
The wheel odometry is well calibrated in translation. The results here are not usable.

Another interesting thing is that after a crash (unknown reason) the camera restarted with apriori good values and then diverge:
[image]

So I have multiple questions:
- Is it normal?
- Is it due to temperature?
- Is it due to dynamic calibration?
- What can I do to make it more stable?
- Some month ago the values where usable, is it due to firmware update?
- What can we expect about the stability of the values?

BriceRenaudeau, 2019-10-15:
I tried to downgrade the librealsense and the ros-wrapper.
It appeared that the result are similars, hence it's not a probleme of version.

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

@BriceRenaudeau

We are always working on our algorithm and making updates as you may know based on updates to the latest librealsense and outlined via our release notes.

With that being said, has anything changed in regards to the environment or lighting or anything else in the operations of your robot device?

Were you able to update to the latest librealsense 2.30.0?  Did this change the behavior?

BriceRenaudeau, 2019-12-02:
Hello, 
I tried with the latest lib (2.30) but the result is the same.
The linear velocity is not stable.

chowjf, 2019-12-05:
@BriceRenaudeau 
Hi there, other than the velocity drifting, do you experience a constant odometry (pose) drifting over distance if compare to wheel odometry?

BriceRenaudeau, 2019-12-05:
Yes, the drift in the velocity can be seen in the pose. 
Here is the linear vel and the distance traveled by the wheels and by the T265
[image]
Apart from the delayed start (the camera is somehow waiting for enough features), we can see that the camera is traveling less distance.

schmidtp1, 2019-12-09:
Hi @BriceRenaudeau, thanks for providing the results of your experiments.
You mention that the wheel odometry is well calibrated in translation, are the extrinsics, i.e. the relative transformation between camera and odometry also precisely calibrated? A misalignment could lead to a scale error. (It could be interesting to look at the norm of the velocities to assess this.)
It is curious to see that it doesn't seem to be a pure scale issue but that there is a nonlinear effect since the velocities match better at lower velocities. Would you have the chance to record an additional external measurement as reference, for example from an external tracking system?
I'd also like to understand if you are fusing wheel odometry at all *inside* the camera, i.e. sending it  to the T265, or just "externally" inside your own Kalman filter? In the first case, does the estimated velocity of the camera include wheel odometry (fusion) in this comparison?

BriceRenaudeau, 2019-12-10:
I am not using the wheel odometry in the camera, I prefer to use only external EKF to avoid data incest.
I already checked the linear/Y velocity, it is 0 all the time (except while rotating).
I don't have an external tracking system but I have laser scans and amcl running on the robot. It shows that the wheel velocity is correct.

RealSenseSupport, 2020-09-11:
Thank you for highlighting an issue with precision on the T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-16 by RealSenseSupport