realsenseai/librealsense #5006: udev rules error gone with manual installation but Extensions type is unhandled

cassie101, 2019-10-08:
Camera Model: D435i
Operating System: Ubuntu 18.04.3 LTS
Kernel Version (Linux Only): Linux wlan 4.15.0-65-generic

Hi, I was running `realsense-viewer` and initially faced `out of date udev rules` following the installation guide [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages). I then tried to install via [manual installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) and the `udev rules error` is gone, however I still face `08/10 16:27:29,523 WARNING [140050783844096] (ds5-device.cpp:502) The IMU sensor is undefined for PID ad6`.

Beside the above issue, when I click 'record' in realsense-viewer, I also faced the error message below. Is this how it is supposed to be? If not, how could I fix it? Thanks.
```
 08/10 16:28:03,396 WARNING [140050679453440] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 08/10 16:28:07,526 ERROR [140050662668032] (record_device.cpp:210) Failed to downcast snapshot of type librealsense::debug_interface
 08/10 16:28:07,527 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Motion
 08/10 16:28:07,527 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Video
 08/10 16:28:07,527 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Roi
 08/10 16:28:07,527 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Depth Frame
 08/10 16:28:07,527 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Advanced Mode
 08/10 16:28:07,527 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Video Profile
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Disparity Frame
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Motion Profile
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Pose Frame
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Pose Profile
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Tm2
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Software Device
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Software Sensor
 08/10 16:28:07,528 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Decimation Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Threshold Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Disparity Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Spatial Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Temporal Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Hole Filling Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Zero Order Filter
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Pose
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Pose Sensor
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Wheel Odometer
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Global Timer
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Updatable
 08/10 16:28:07,529 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Update Device
 08/10 16:28:07,530 WARNING [140050662668032] (record_device.cpp:255) Extensions type is unhandled: Tm2 Sensor
```

kjacks21, 2019-10-08:
Potentially related to #4990. @cassie101 have you tried the new firmware update announced in #5003? I haven't had a chance to yet, but will later today.

ev-mp, 2019-10-10:
@cassie101 hello, the warning 
>08/10 16:27:29,523 WARNING [140050783844096] (ds5-device.cpp:502) The IMU sensor is undefined for PID ad6

says two things:
1. The D435i camera is identified as USB2 device.
2. the firmware that you're using is rather old one, and it does not support D435i properly in USB2 mode. (no RGB and no IMU sensors).
You should run `realsense-viewer` and upgrade the firmware to the recommended version. Then you can review the USB cabling/port to resolve a possible improper USB connection.

The logs regarding the recordings are expected and not indicative of misbehavior.

cassie101, 2019-10-10:
@ev-mp Hi, thank you, I updated the firmware and use a usb3, the error message is gone.
Do you know what does the following error mean?
> (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).

ev-mp, 2019-10-29:
@cassie101 , sorry for a delayed response
The error indicates that the D435i IMU is not calibrated, which is the default. So it can be safely disregarded.
It is possible to calibrate IMU by yourself using [SDK-provided tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration#rs-imu-calibration-tool) and remove the warning, but since the factory IMU calibration is fairly good very few customers opted to do so.

closed 2019-11-01 by cassie101