realsenseai/librealsense #5007: I think, the initial develop setting is wrong.

devLupin, 2019-10-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | Latest | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  Latest                           | 
| Language                            |  C++/C#                       | 
| Segment			|  others                           | 

### Issue Description
<question>
[image]

I'm super rookie developer. Please understand even this low question.

I added the following path to the (project properties->C/C++->Additional Include Directories).
C:\Users\devLupin\Intel_Realsense\librealsense-2.25.0\librealsense-2.25.0\include

I think the initial development setup is wrong. Can you point it out?
Thank you.

devLupin, 2019-10-08:
<Attachment link(VS capture.jpg)>
https://user-images.githubusercontent.com/33558083/66381359-209d9100-e9f4-11e9-9b52-71a8eaeb799e.JPG

stefan-enway, 2019-10-08:
besides from pasting an image, what is your question? help how to code?

devLupin, 2019-10-10:
@stefan-enway 

Umm and Um,,,
As you can see in the picture, the problem is caused by duplicated functions in the header files. Tell me if the initial setting is the same as I wrote in the first posting.

If not, please tell me the initial settings. Thank you.

devLupin, 2019-10-10:
I've solved it because I keep searching Google.

Thank you stefan:)

ev-mp, 2019-10-22:
@devLupin , the issue seem to be resolved so I'm closing it now.
Feel free to reopen if it was not the case.

closed 2019-10-22 by ev-mp