realsenseai/librealsense #5008: Saving file names in .txt file

aalarcon96, 2019-10-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   |05.11.11.100| 
| Operating System & Version |   Win (10)   | 
| Platform                                 | PC |
| SDK Version                            |  { 2.25.0.993 }                          | 
| Language                            |  {C/C#                        | 

### Issue Description

Hello, 

In the align-advanced example, I have added code to save .png color and depth files for each frame as well as .csv files measuring the depth for each pixel of those .png pictures. I would like to create a text file containing the file names of each of these .png and .csv files. Can anyone help me figure out the code for this? I am a bit new to C++. 

Thank you.

RealSenseCustomerSupport, 2019-10-17:
----------------------------------------------

one way you may consider is to use DOS/Powershell command

dir *.png  -Name >list.txt
dir *.csv -Name  >>list.txt

you can create a batch file to execute it

CarltonBranch, 2019-10-21:
Hi, I looked at the align-advanced.cpp file and I'm not sure that I understand what you're trying to do. Could you be more specific?

aalarcon96, 2019-10-23:
So I added code in the while (app) part of the code to save the color and depth images in .png format. I need a way to include the names of those picture files into a .txt file. I tried using this code:

` 
int index;
 index++;
        // save color and depth images
        std::stringstream png_file;
         png_file << "rs-save-to-disk-output-" << other_frame.get_profile().stream_name() << std::to_string(index) << ".png";
         stbi_write_png(png_file.str().c_str(), other_frame.get_width(), other_frame.get_height(), other_frame.get_bytes_per_pixel(), other_frame.get_data(), other_frame.get_stride_in_bytes());
         std::cout << "Saved " << png_file.str() << std::endl;
        
         rs2::video_frame vf = c.process(aligned_depth_frame);
         std::stringstream png_file1;
         png_file1 << "rs-save-to-disk-output-" << vf.get_profile().stream_name() << std::to_string(index) << ".png";
         stbi_write_png(png_file1.str().c_str(), vf.get_width(), vf.get_height(), vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());
         std::cout << "Saved " << png_file1.str() << std::endl;

// code to save file names in .txt
 std::ofstream file;
 file.open("color.txt");
 file << "rs-save-to-disk-output-Color" << std::to_string(index) << ".png \n";
 file.close();
`

But it only saves the very last file name. I hope this makes sense. Basically, I get 
rs-save-to-disk-output-color-1.png, rs-save-to-disk-output-color-2.png, rs-save-to-disk-output-color-3.png... etc. and I want to save those names into one txt file, but I only get the last one. Is there a way to save all of the names into one text file?

Thank you!

aalarcon96, 2019-10-23:
Also, I switched to MacOs Mojave if that makes a difference.

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

it looks you overwrite the file name
you can check stackoverflow for more suggestion
https://stackoverflow.com/questions/2393345/how-to-append-text-to-a-text-file-in-c

closed 2019-11-20 by RealSenseCustomerSupport