realsenseai/librealsense #5009: Trying to merge D415 depth frames knowing their pose data from T265

AJMena, 2019-10-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10}   | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.25.0.993 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description
I have two ".bag" files. One from D415 the other from T265. Both recorded almost at the same time. I load both and synchronize the frames, so I get the depth, colour and pose at the same time. Now I select two frame and try to merge them together. To that purpose I calculate the transformation matrix using the function quat2mat from example.hpp. I apply this matrix to every point in the depth frame. Then I export the results and check them in the program CloudCompare. But the result doesn’t match.

I verified every step several times but no matter what the results don't match. I checked the transformation operation several times with Cloudcompare and I get the same result. I think maybe I need to do something with the Quaternion before obtaining the transformation matrix.

My actual code is this:

```
inline void quat2mat(rs2_quaternion& q, GLfloat H[16])  // to column-major matrix
{
    H[0] = 1 - 2*q.y*q.y - 2*q.z*q.z; H[4] = 2*q.x*q.y - 2*q.z*q.w;     H[8] = 2*q.x*q.z + 2*q.y*q.w;     H[12] = 0.0f;
    H[1] = 2*q.x*q.y + 2*q.z*q.w;     H[5] = 1 - 2*q.x*q.x - 2*q.z*q.z; H[9] = 2*q.y*q.z - 2*q.x*q.w;     H[13] = 0.0f;
    H[2] = 2*q.x*q.z - 2*q.y*q.w;     H[6] = 2*q.y*q.z + 2*q.x*q.w;     H[10] = 1 - 2*q.x*q.x - 2*q.y*q.y; H[14] = 0.0f;
    H[3] = 0.0f;                      H[7] = 0.0f;                      H[11] = 0.0f;                      H[15] = 1.0f;
}

inline void VertexTransformation(GLfloat t[16], GLfloat v[3], GLfloat r[3])
{
	r[0] = t[0] * v[0] + t[1] * v[1] + t[2]  * v[2] + t[12];
	r[1] = t[4] * v[0] + t[5] * v[1] + t[6]  * v[2] + t[13];
	r[2] = t[8] * v[0] + t[9] * v[1] + t[10] * v[2] + t[14];
}

void AddFrameToCloud(std::vector<Point3D>& cloud, rs2::frame& depth, rs2::frame& color, rs2::frame& pose)
{
	GLfloat vertex[3];
	GLfloat result[3];
	GLfloat transformation[16];

	rs2::pointcloud pc;
	pc.map_to(color);
	rs2::points points = pc.calculate(depth);

	auto vertices = points.get_vertices();              // get vertices
	auto tex_coords = points.get_texture_coordinates(); // and texture coordinates

	// Cast the frame to pose_frame and get its data
	auto pose_data = pose.as<rs2::pose_frame>().get_pose_data();
	rs2_quaternion Quatrnion = pose_data.rotation;

	quat2mat(Quatrnion, transformation);
	transformation[12] = pose_data.translation.x;
	transformation[13] = pose_data.translation.y;
	transformation[14] = pose_data.translation.z;

	for (int i = 0; i < points.size(); i++)
	{
		if (vertices[i].z && -10 < vertices[i].z && vertices[i].z < 10
			&& tex_coords[i].u >= 0 && tex_coords[i].v >= 0 
			&& tex_coords[i].u <= 1 && tex_coords[i].v <= 1)
		{
			vertex[0] = vertices[i].x;
			vertex[1] = vertices[i].y;
			vertex[2] = vertices[i].z;

			VertexTransformation(transformation, vertex, result);

			Point3D Point = Point3D(result[0], result[1], result[2]);

			cloud.push_back(Point);
		}
	}
}
```

Any idea of what am I doing wrong?

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Hi @AJMena - ev-mp referenced another github ticket that is similar.  I believe the information there would also be valuable here as well.

https://github.com/IntelRealSense/librealsense/issues/5085

AJMena, 2019-11-22:
I kind of solved it. The results still don't match perfectly. There's +-2 cm error overlap between frames. I think that is the precision error of the D415. Could it be that?
The code actually looks like this:

```
void TransformCloud(std::vector<Point3D>& cloud, rs2::frame& pose, float H_t265_d400[16])
{
	GLfloat tempresult0[3], tempresult1[3]; // We need to use a second variable to not mess overwriting.
	GLfloat transformation[16];

	// Cast the frame to pose_frame and get its data
	auto pose_data = pose.as<rs2::pose_frame>().get_pose_data();
	rs2_quaternion Quaternion = pose_data.rotation;

	// We checked that we needed to invert de sign.
	Quaternion.x = -Quaternion.x;
	Quaternion.y = -Quaternion.y;
	Quaternion.z = -Quaternion.z;

	quat2mat(Quaternion, transformation);
	transformation[12] = pose_data.translation.x;
	transformation[13] = pose_data.translation.y;
	transformation[14] = pose_data.translation.z;

	int size = cloud.size();
	for (int i = 0; i < size; i++)
	{
		tempresult0[0] = cloud[i].X;
		tempresult0[1] = cloud[i].Y;
		tempresult0[2] = cloud[i].Z;

		// We have to apply H_t265_d400 before the transformation
		VertexTransformation(H_t265_d400, tempresult0, tempresult1);
		VertexTransformation(transformation, tempresult1, tempresult0);

		cloud[i].X = tempresult0[0];
		cloud[i].Y = tempresult0[1];
		cloud[i].Z = tempresult0[2];
	}
}
```

Check this example to know what is the H_t265_d400:
https://github.com/IntelRealSense/librealsense/tree/master/examples/tracking-and-depth
The VertexTransformation and quat2mat functions are the same as before.
The rest I think is self-explanatory.

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

Hi @AJMena

Glad to see that you were able to get the setup working.  I would expect some error as well.  Let me know if we can go ahead and close this out.

closed 2020-02-03 by RealSenseCustomerSupport