realsenseai/librealsense #5010: Need help in Multiple Camera Realsense D435 calibration

dk67604, 2019-10-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Currently we tried box dimensioner example, it's not giving ver accurate result when the transformation happens using the calibration from local camera coordinate to the global coordinate. The setup has three camera. I have following questions, please help me :
1) Does the each camera has to be in same orientation, currently we have the positioned horizontally, flipped horizontally, vertically placed.
2) Commom field of view is below 40 cm for all the three cameras. 
I am looking for a method which can provide robust calibration.

MartyG-RealSense, 2019-10-08:
Is this a follow-up to your previous case about box-dimensioner, please?  Did the advice given in that case help?

https://github.com/IntelRealSense/librealsense/issues/4715

dk67604, 2019-10-08:
This is not follow-up on Issue #4715, We are able to calibrate with success and but when using TranformationPose(R|t) matrix the camera which is positioned vertically has a very inaccurate result. Please suggest some robust in calibrating multiple realsense camera. Please let me know if you need more input from my side.

MartyG-RealSense, 2019-10-08:
A tool that Intel themselves have used for multi camera calibration of 400 Series cameras is Vicalib.  
It uses a board that you show to each of the cameras in turn., and it establishes overlapping regions to then minimize the pose of each of those together.

https://github.com/arpg/vicalib

dk67604, 2019-10-08:
Thanks @MartyG-RealSense  I can try this tool, is tool only availaible to install through source code?

MartyG-RealSense, 2019-10-08:
As far as I know you have to compile Vicaib yourself. Intel also sell a professional calibration system that can calibrate multiple cameras and do so through a remote server, though it costs 1500 dollars due to the cost of manufacturing the large calibration chessboard.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

dk67604, 2019-10-10:
@MartyG-RealSense Is the calibration system can offer calibration for multiple cameras looking to a common scene. The whole arrangement is static? If yes would like to get in touch with the Intel Support team to understand and order it.

MartyG-RealSense, 2019-10-10:
I do not know enough about that aspect of the tool to comment on it.  I will link this discussion to Intel's @agrunnet  who may be well equipped to answer.

dk67604, 2019-10-11:
That would great! Thanks looking forward to connecting.

dk67604, 2019-10-14:
@MartyG-RealSense Would be great if you share vicalib command line usage when 3 cameras are connected. Thanks again for all the support!

dk67604, 2019-10-14:
@agrunnet I'm looking forward to talking with you.

MartyG-RealSense, 2019-10-14:
@dk67604  You're very welcome.  I've never used Vicalib myself though, so have no insight about its detailed operation.

dk67604, 2019-10-14:
@MartyG-RealSense Is there anyway I can connect with some one. Please help me with this, we are trying to solve this problem from last 1 month.

MartyG-RealSense, 2019-10-14:
I have no influence over when Intel RealSense team members respond, unfortunately.  They are required to visit each case at least once to ask if help is still required, and close the case if the problem is resolved or nobody responds that they want to keep it open.  I do not know exactly when they will comment though.

I re-read your case from the start to try to get new insight.  I am not sure that special calibration is required.  As the D435 that you are using does not have an IMU (and so does not have a gyro component that responds to tilt), the direction that the camera is pointed in should not make a difference.  

Looking over the instructions in the box-dimensioner example, I don't believe that a calibration solution like Vicalib is going to help.  Box-dimensioner seems to use its own in-built calibration system to analyze the chessboard that the object being measured is placed on.

So there is an important distinction to make:

1.  You could use the Ubuntu version of the Dynamic Calibrator tool to calibrate each camera **individually** to help ensure that you are going to get a good-quality image.

2.  But **then** the code in the box-dimensioner example carries out its own kind of calibration of the multiple cameras by observing the chessboard, and this enables it to get good measurements of the object on the chessboard.

Looking at the main instructions page for box-dimensioner, it seems to say that no object should be on the chessboard when you start the program.  When the program starts, it should automatically carry out its multiple-camera calibration.   An object should not be placed onto the chessboard until the program asks you to.

If you are using your own custom size of chessboard then the program's chessboard parameters need to be edited so that you get correct results.  Also, make sure that the chessboard is in the view of **all** the cameras.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam#workflow

dk67604, 2019-10-14:
Thanks @MartyG-RealSense I appreciate your efforts for instructions. I'll definitely be going to try Dynamic Callibrator tool, and then the box-dimensioner example.

dk67604, 2019-10-23:
@MartyG-RealSense I tried to Ubuntu version of the official Intel 'Dynamic Calibrator' tool, tried a couple of times and followed the steps shown in demo. Everytime I'm getting this error.[
[image]
](url)

MartyG-RealSense, 2019-10-23:
Are you using a USB 3 port with the tool, please?

dk67604, 2019-10-23:
@MartyG-RealSense Yes I make sure that I'm using USB 3.0 port. I can try different machine

RealSenseCustomerSupport, 2019-11-05:
----------------------------------------------

Hi dk67604

Can you try the Dynamic Calibrator tool under Windows 10 instead of Ubuntu? I just want to make sure your operation of the calibration is correct. This is to help you build the experience about how to use the tool.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

dk67604, 2019-11-07:
I'll try on Windows 10

closed 2019-12-11 by RealSenseCustomerSupport